OrcaSlicer/OrcaSlicer #7747: Creality K2 plus device tab shows 404 Not Found instead of web interface

wreutz, 2024-12-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

Sonoma 14.7.2

### Additional system information

Apple macStudio M1 Max 

### Printer

Crealiity K2 plus combo

### How to reproduce

1. click connection next to printer
2. select Host Type "Creality Print"
3. enter IP address
4. click "Test" button -> successful
5. switch to "Device" tab

### Actual results

 shows "Error 404 Not Found" Nginx webpage

### Expected results

show Creality UI as in Creality Print 5.1.6

### Project file & Debug log uploads

[debug_Sun_Dec_15_01_58_11_4982.log.0.zip](https://github.com/user-attachments/files/18138552/debug_Sun_Dec_15_01_58_11_4982.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Hammerfest, 2024-12-15:
Add the port.
:4408

00raq00, 2024-12-15:
> Add the port. :4408

Nope 4408 is not a prot for creality interface

dwaite, 2024-12-21:
I believe someone would need to know what port to use for the creality interface to fix this issue.

guyrules, 2025-01-11:
I can get to my K2 camera with :8000

kanocz, 2025-01-19:
tcpdump shows that CrealityPrint is connecting to port :9999 and then (unencrypted) websocket connection is established

dwaite, 2025-01-21:
The Creality Print device tab is natively rendered in the slicer based on the 9999 interface; there is no current web interface to display other than fluidd. 

Unfortunately, there's no support like AFC to remote-administer the CFS or do mapping for prints in the installed fluidd currently. So starting a multicolor print needs to be done either on the touch screen, via CP or the CC app

closed 2025-04-29 by github-actions[bot]