OrcaSlicer/OrcaSlicer #7748: Keyboard shortcut for object's printability and toggling object printability on multiple selected objects at once.

MAJCcustoms, 2024-12-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I often set up huge build plates for my farm to be as efficient as possible. Sometimes I only want to print a few items of that plate and I want to reslice to take advantage of the slicer's speed control (in other words using Object Exclusion in klipper or the likes is not an option). I have a saved project for each item I usually print with different plates for different colors so I don't want to make changes like moving stuff around each time I just dont want something on the saved plate to be printed. This usually means I have to manually select each object to disable in the "objects" section. There doesn't seem to be a hotkey for this or even the ability to do multiple objects at once which would save a huge amount off time for my use case.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Hotkey for object disabling and/or disable multiple selected objects at once.

tlhintoq, 2024-12-16:
[image]

MAJCcustoms, 2024-12-17:
A tick box in the header would be a nice cherry on top for more usability.

fritzw, 2024-12-19:
You can already select multiple objects, then right click one them (either in the object list or in the 3D view), and click "Printable". This will toggle the selected items on or off. It even has a checkbox to show the current state in the menu.

tlhintoq, 2024-12-19:
> You can already select multiple objects, then right click one them (either in the object list or in the 3D view), and click "Printable". This will toggle the selected items on or off. It even has a checkbox to show the current state in the menu.

nice... I guess I never tried that 42nd permutation of possible UI options.  
Thanks for sharing the Secret Squirrel handshake.

MAJCcustoms, 2024-12-20:
> You can already select multiple objects, then right click one them (either in the object list or in the 3D view), and click "Printable". This will toggle the selected items on or off. It even has a checkbox to show the current state in the menu.

That's nice. Thanks. I knew I used to be able to do it somehow when I used PrusaSlicer but couldn't remember and didn't see anything while googling. I would still like a hot key for this, though.