OrcaSlicer/OrcaSlicer #7749: "Unsupport print file path" if there is a plus ( + ) in the file name

tlhintoq, 2024-12-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev nightly 

### Operating System (OS)

Windows

### OS Version

Win10proX65

### Additional system information

Win10pro

### Printer

BambuLab

### How to reproduce

Pretty common from Makerworld (maybe others) where object names have a plus in the file name in place of a space.

It looks like the BambuLab machines (X1C and P1S) don't like that.
If I send the same job but rename on send and take out the plus its happy

[Phone+stand.zip](https://github.com/user-attachments/files/18138820/Phone%2Bstand.zip)

### Actual results

Error and fail
[image]

### Expected results

Maybe automatically replace invalid characters with an understore

### Project file & Debug log uploads

Any file with a plus in the name will do it
[Phone+stand.zip](https://github.com/user-attachments/files/18138827/Phone%2Bstand.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

any file with a plus in the name will do it

Noisyfox, 2024-12-15:
Did you send with Lan mode or over BBL's cloud?

tlhintoq, 2024-12-16:
> Did you send with Lan mode or over BBL's cloud?

BBL cloud

tlhintoq, 2024-12-16:
After some more working with this I realize its the file name, not the model name.

So if I have `Widget.3mf` with plate `Thing` then the file is uploaded as `widget_thing.gcode`.
The problem comes in when you grab a file off Makerworld such as `Some+Tool.stl` and drag that to a new Orca project.
There is no file name until you save.  If you just drag and print you get `Some+Tool_PlateName.gcode` and the plus is illegal even though its the default naming convention on Makerworld.

Noisyfox, 2024-12-18:
Weird... I couldn't reproduce this issue on my machine.

closed 2025-03-27 by github-actions[bot]