OrcaSlicer/OrcaSlicer #7751: Prime volume field missing

pgifford, 2024-12-15:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev (14 Dec 2024)

### Operating System (OS)

Windows

### OS Version

10

### Printer

A1 Mini, P1P

### How to reproduce

1. Create a new project
2. Click on Prepare
3. Click on Multimaterial tab

### Actual results

Prime volume field is not present
[image]

### Expected results

Prime volume field is present (screenshot from v2.2.0 stable)
[image]

### Project file & Debug log uploads

Not relevant, the Prime volume field is missing from the most recent Windows v2.3.0-dev portable.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Prime volume shows up in search and if selected the UI will switch to the Multimaterial tab.
[image]

igiannakas, 2024-12-18:
This has been fixed in PR #7789

Can you please test and confirm? thank you!!

Noisyfox, 2024-12-21:
Try #7808

closed 2024-12-21 by SoftFever

aaron-crocker, 2025-01-26:
Has anyone found a solution to this yet.  Prime volume is missing from my most recent version of Orca slicer.  It shows up in the search field and even snaps over to the Multimaterial tab.  But there is no prime volume option.  :(

igiannakas, 2025-01-26:
Spot on. It won’t use priming if the prime tower is used for purging :)

PR #7808 (2024-12-21): Fix Prime volume missing on BBL printers
Issue is caused by #7629, which overwriten the profile with BBL's, which has some our unique properties missing. This PR brings them back.

Fixes #7751