OrcaSlicer/OrcaSlicer #7760: Reorder Filaments

Investegate, 2024-12-15:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I wanted to edit a 3MF-file created in OrcaSlicer for Bambu X1 Carbon. It has 7 Filaments selected. If i change it now to my Prusa XL Multitool printer, it removes Filament 6&7. But in this case i need the multicolor painting from Filament 7 and there is no way to change the order of the filaments. So i lost the multicolor painting...

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

You should have the possibility to change the order of the filaments here:

[image]

tlhintoq, 2024-12-16:
Don't you just select it in the drop down?
If you want that blue from slot 6 in the first slot, go to the first slot drop down and select it there.
The order of these slots is the printing order, so keep that in mind as you change them.

Investegate, 2024-12-16:
> Don't you just select it in the drop down? If you want that blue from slot 6 in the first slot, go to the first slot drop down and select it there. The order of these slots is the printing order, so keep that in mind as you change them.

No thats not possible. I can only change the filament type there. The multicolor painting stays at Slot 6. I cant move it to an other slot.

fritzw, 2024-12-19:
Currently there is a 1 to 1 correspondence between model colors and printer filament slots.

I think it would be nice if the multicolor painting would be more independent from the filament selection and the currently selected printer. The color painting information should be preserved on the model regardless of which printer is selected. If the printer has fewer filament slots than there are colors on the model, it should be possible to assign the same filament slot to multiple model color slots, instead of losing the color information.

This would make it much easier to design multicolor models that work on different printers without having to redo the painting for each filament count. Also, this would make it possible to paint models freely (for uploading to model sharing websites) without having to configure a multi-tool printer that you don't own. The person who prints it can then select which of the painted colors they want to use and which not.

However, I'm not sure how this change could be incorporated in the 3mf file format and how much it would break compatibility with PrusaSlicer/BambuStudio when exchanging 3mf files between programs.

tlhintoq, 2024-12-19:
I have to admit - It sure would be nice to go from a X1C+AMS to a P1S+AMS, both loaded with the same colors and not loose painting.

SoongJr, 2026-02-27:
Oh, just found a workaround: upload gcode to the printer, then start the print from the printer's display, it then asks me to confirm or change assignment of loaded filament