OrcaSlicer/OrcaSlicer #7762: Combination of "Only one wall on top surfaces", line width and Arachne wall generation leads to messed up additional layer

bioxz, 2024-12-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Arch Linux (latest)

### Printer

Creality Ender-3 V3 SE

### How to reproduce

Create a new project and apply the quality settings as in the provided project, especially enable

- _Only one wall on top surface_ 1
- _Wall generator_ Arachne
- Line width (relevant might be: Outer wall 0.2mm, Inner wall 0.45mm)

and slice the object.

I don't know exactly which line width setting causes the issue, but my default and inner wall are more than twice as thick as the outer wall.

### Actual results

[image]

The object gets sliced, but there is no top surface. Instead an additional layer gets created including an unprintable sausage looking thing. The gcode generated from this additional layer has too much extrusion for my printer and fails the print.

[image]

This does not only happen with the supplied model, but also with many more and thus has to be caused by the combination of my settings. Disabling either _Only one wall on top surface_  or setting _Wall generator_ to Classic leads to the generation of a normal top surface. Leaving the _Outer wall_ at 0.2mm and setting the _Outer wall_ to 0.37mm or less also fixes the issue.

Source for object file: https://www.printables.com/model/1096828-knitted-snowman-in-parts

### Expected results

The slicer should use the Arachne wall generator and create a top surface with only one wall using the defined _Outer wall_ line width, while the other layers have additional walls using the defined _Inner wall_ line width. There should be no additional layer on top the object.

### Project file & Debug log uploads

[snowman_scarf_red.zip](https://github.com/user-attachments/files/18142822/snowman_scarf_red.zip)

Source for object file: https://www.printables.com/model/1096828-knitted-snowman-in-parts

### Checklist of files to include

- [ ] Log file
- [X] Project file

MxBrnr, 2024-12-16:
I'm fairly certain the issue here is trying to use a line width that is half the size of the nozzle, which is generally a no-no. Changing just the outer wall line width to anything that is 60% or greater of nozzle diameter instantly fixes the issue.

bioxz, 2024-12-16:
> I'm fairly certain the issue here is trying to use a line width that is half the size of the nozzle, which is generally a no-no. Changing just the outer wall line width to anything that is 60% or greater of nozzle diameter instantly fixes the issue.

No, it actually doesn't fix the issue. Outer/Inner values of 0.2mm/0.4mm, 0.25mm/0.5mm or 0.3mm/0.6mm all cause the same issue.  If I can print those lines should be a different thing. Maybe a better example if you can't print 0.2mm: I can also reproduce this with 60%/120% line width for Outer/Inner.

[image]
[image]

The slicer is seemingly producing these results due to the difference between inner and outer and the wall generator selection. Somehow _Only one wall on top surfaces_ breaks the algorithm with such a big difference. The absolute numbers do not matter, the difference does.

[image]
[image]

igiannakas, 2024-12-18:
There's not much that can be done about this, beyond adding a check in the UI to disable this feature when such a large variation in extrusion size is detected. That is because of the way one wall on top surfaces is implemented. This is probably also reproducible in Prusa slicer (we've updated in 2.2 the algorithm to mirror prusa's implementation as it fixed several issues with the previous approach).

https://github.com/SoftFever/OrcaSlicer/pull/6236

vovodroid, 2025-02-03:
I don't see this issue in latest Orca:

[image]

igiannakas, 2025-02-03:
Try their line widths , it caused it for me too when I tested it I think.

vovodroid, 2025-02-03:
I've open original project with **bioxz** settings.

igiannakas, 2025-02-03:
Huh yes indeed it seems fixed! Then @bioxz can you confirm and close the issue please?

bioxz, 2025-02-09:
Cannot reproduce the bug in build from git, seems like it got accidentally fixed, great!

closed 2025-02-09 by bioxz