OrcaSlicer/OrcaSlicer #7764: "Maximum acceleration for travel" grayed out in Ender-3 V2 Printer Preset

Ergonomicmike, 2024-12-16:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev

### Operating System (OS)

Windows

### OS Version

Win10

### Printer

Creality Ender 3 V2

### How to reproduce

Printer>Creality Ender-3 V2 0.4 nozzle>Edit Preset>Motionability 

### Actual results

The setting for Maximum acceleration for travel is grayed out, cannot be changed by the user, unlike all the other acceleration limits.
[image]

### Expected results

Shouldn't I be able to set this too?

### Project file & Debug log uploads

[Grayed Out Travel Accel bug report.zip](https://github.com/user-attachments/files/18152003/Grayed.Out.Travel.Accel.bug.report.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

Admittedly a small problem, and probably not important for normal Ender users. But here I'm using Klipper.

As a work around, I hacked the fdm_creality_common.json file, found in OcraSlicer/system/Creality

[image]

Ergonomicmike, 2024-12-16:
I added 
```
"machine_max_acceleration_travel": [
        "5000"
	],
```
to the file fdm_machine_common.json, thinking that this change would wake up the grayed out limit.

No change.

I'm now thinking it's a problem in the UI? (Since no hint pops up on mouse hover.)

Ergonomicmike, 2024-12-16:
To test to see if this issue follows the printer or stays with the UI, I just now installed the generic Klipper printer preset.

"Maximum acceleration for travel" is still grayed out. (Albeit now at 20000."

MxBrnr, 2024-12-16:
If you read the tooltip you will see that setting only applies to Marlin 2 firmware; you are running Klipper, that is why it is grayed out.

Ergonomicmike, 2024-12-16:
@MxBrnr 

Tnx. No tooltip for me, but based on what you're saying, that's because I've set Klipper as my G-code flavor. 

So I clicked over to Marlin 2, and I can now see the tool tip about M204 T. 

So since Accleration for Travel is grayed out for Klipper, does this mean that the value shown doesn't matter, isn't a limit in Klipper? (If it weren't a limit, then why is it set to 20000 for the generic Klipper printer preset?)

Another way of asking the same question is: Does machine_max_acceleration_travel set a limit on a Klipper machine?

But okay, I see this issue about the G-code being grayed out isn't a bug. I'll close this soon.

Ergonomicmike, 2024-12-16:
I suppose this is all moot if I uncheck "Emit limits to G-code." I thought that putting the limits in the G-code would help KlipperScreen estimates. Pros/Cons?

Ergonomicmike, 2024-12-16:
[image]
Well, now, if none of these limits will be sent when using Klipper, then why isn't this option grayed out for Klipper too?

Ergonomicmike, 2024-12-16:
So then I loaded the generic Klipper printer preset and Look! "Emit limits to G-code" is hightlighted!

Very confusing.

[image]

MxBrnr, 2024-12-16:
> ![Oops](https://private-user-images.githubusercontent.com/15025888/396196567-ccedeaf6-a352-4f60-9b7d-6e327ef5bcb7.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQzNjkzNDMsIm5iZiI6MTczNDM2OTA0MywicGF0aCI6Ii8xNTAyNTg4OC8zOTYxOTY1NjctY2NlZGVhZjYtYTM1Mi00ZjYwLTliN2QtNmUzMjdlZjViY2I3LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEyMTYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMjE2VDE3MTA0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTI3NmQ2ZWQ5ZDA0YmU1YWU3NDdiNjQ5YTFhMjA0ZTc2Yzk5NWMyY2NhZGUwOTUwMTc5ZGNmZjUyMDRkNzE2NTkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.oCSIqc8YnCqlrEDP6wIFPOQrL2Od_AEuhyp4336VGn4) Well, now, if none of these limits will be sent when using Klipper, then why isn't this option grayed out for Klipper too?

I probably should be grayed-out, or just invisible when Klipper firmware is selected; in the same way, `Maximum acceleration for travel` should also really just be invisible. Both of these changes would prevent confusion such as has happened here.

Klipper doesn't actually use any hard limits for speeds and accels, so these settings will act as software-only limits, allowing Orca to give warnings and prevent slicing if the accels set in the Process profile exceed those set in the Machine profile. This leaves it up to the user to ensure that any speed/accel settings used do not exceed the machines physical capabilities.

Ergonomicmike, 2024-12-16:
To test the Tooltip statement that "Emit limits to G-code" won't be including in the G-code if using Klipper, I sliced a Benchy with Emit selected, and then unselected. (Everything else unchanged.)

My expectation was, "If emit limits doesn't emit limits when using Klipper, then the G-code should be the same with the checkbox checked or unchecked." 

But they're not. 

The Model print times are identical. So that confirms that there's no limit change.

But, surprisingly, the G-ccode files are not the same. 

Attached is the Project File from Winmerge, comparing the two.

[Benchy.Emit.NoEmit.WinmergeProject.zip](https://github.com/user-attachments/files/18154775/Benchy.Emit.NoEmit.WinmergeProject.zip)

Ergonomicmike, 2024-12-16:
> Klipper doesn't actually use any hard limits for speeds and accels, so these settings will act as software-only limits, allowing Orca to give warnings and prevent slicing if the accels set in the [...]

From my Klipper printer.cfg
[image]

MxBrnr, 2024-12-22:
These are not "hard" limits, they are just defaults. Klipper only uses a single command, `SET_VELOCITY`, for setting speed/accel at any given moment (including from the slicer), and this command overrides those set in the firmware. When you manually adjust these parameters through Mainsail/Fluidd, it does the same. The firmware defaults only act as maximums *unless otherwise specified.*

closed 2025-06-29 by github-actions[bot]