OrcaSlicer/OrcaSlicer #7768: I hope to add a separate control function for the first layer traffic

lanshui777, 2024-12-17:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Using rough and smooth printing boards with different gaps to fill can result in different results for the same flow configuration on the underlying printing surface. When I adjust the smooth printing board to the appropriate parameters and then use a rough printing board, the bottom surface needs to increase the flow rate to achieve the same surface effect, but this also leads to a change in the flow rate of the later layers.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I hope to increase the independent control function of the first layer flow to adapt to different printing board materials and reduce the number of repeated parameter adjustments

VigilantGardener, 2024-12-19:
Does this enhancement also cover what you're looking to accomplish?

https://github.com/SoftFever/OrcaSlicer/issues/7684

tiziano149, 2024-12-27:
> Does this enhancement also cover what you're looking to accomplish?
> 
> #7684

This actually does sound like the mentioned enhancement to have individual "per-element" control of Flow ratios.