OrcaSlicer/OrcaSlicer #7771: Tempurature calibration

ging3rb3ardman, 2024-12-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10 Professional

### Additional system information

n/a

### Printer

Bambulabs P1S

### How to reproduce

1.  Go to Calibration
2. Select "Temperature"
3. Input a reasonable start and end temp (My settings were 175 start, 195 end, increments of 5
4.  Click "OK"

### Actual results

This is when I get the following error with a start temp of 350 for some reason even though I had stated something else:

[image]

### Expected results

Should get the proper starting temp to be able to do the temperature calibration test print.

### Project file & Debug log uploads

[Orca Temp calibration bug.zip](https://github.com/user-attachments/files/18163452/Orca.Temp.calibration.bug.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Anything else?

In the works of making a custom printer profile for a rather finnicky filament, so settings might not be dialed in yet.

ging3rb3ardman, 2024-12-17:
Correction, making custom filament profile, not printer.

MxBrnr, 2024-12-17:
Re-read the warning, carefully.
[image]

tlhintoq, 2024-12-19:
In other words...
You have to start hot and end cold, not the other way around.
That's because too hot will still extrude, where too cold will fail.  So if you start too cold you'll never get a tower.

closed 2025-03-27 by github-actions[bot]