OrcaSlicer/OrcaSlicer #7775: [2.3.0] Object labels (M486) cannot be turned off

kamilkrz, 2024-12-17:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Distributor ID: Linux mint  Release: 21.1

### Additional system information

Exists in: [OrcaSlicer_Linux_V2.3.0-dev.AppImage](https://github.com/SoftFever/OrcaSlicer/releases/download/nightly-builds/OrcaSlicer_Linux_V2.3.0-dev.AppImage)

### Printer

Artillery genius pro

### How to reproduce

1. Be sure to turn off "label objects" and "exclude objects"
2. Slice print
3. Check gcode for M486

### Actual results

[image]

### Expected results

M486 obj_name
M486 S0
M486 S -1
Shoudld be not present

### Project file & Debug log uploads

[benchy.zip](https://github.com/user-attachments/files/18170306/benchy.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

tlhintoq, 2025-01-04:
**### One more problem with this bug...**
These `M486` commands are being adding to the Neptune 3 Max jobs - and that's a Marlin machine not Klipper.
This is resulting in the job rebooting the Neptune 3 Max - It just doesn't know what to do with these commands.
[image]

As soon as I comment the `M486` block - the job runs fine.

[image]

Rsoft1976, 2025-03-30:
yes there is, I confirm, the M486 error constantly pops up, it is not enabled in the settings but it shows

PR #9202 (2025-04-13): Fix `M486` still presented even with obj exclusion disabled
Fix #7775
Fix #9200
closed 2025-04-13 by Noisyfox