OrcaSlicer/OrcaSlicer #7783: Prime tower to count as excluded object?

KA009onPC, 2024-12-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I use kamp at the beginning of my prints and I find when I run multi colour prints, kamp only does the object im printing and not the prime tower, is there a way to make the prime tower count as an excluded object so kamp will mesh that area as well? 

Thanks!

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

To be able to have the prime tower count as an object or so it works with kamp or other type bed meshes

VigilantGardener, 2024-12-19:
What benefit does this bring?

I'm not arguing, I don't do much with multicolor and I just want to understand this idea better.

broncosis, 2025-08-30:
I was just about to create a issue for this 
if we are using adaptive bed mesh it does not mesh the area where the prime tower is 
Kamp has now be incorporated in to main stream klipper many more machines use adaptive bed meshing now and this requires the objects to be identified as a exclude able object.

broncosis, 2025-08-31:
While I do like this option but it doesn't work  
 
 When your start gcode is all wrapped into a macro like we have to do in our tool changers  
 
 
 
 
 
 
 
 
 
 
 
 lijgame  left a comment  (SoftFever/OrcaSlicer#7783) 
 
 
 I was just about to create a issue for this if we are using adaptive bed mesh it does not mesh the area where the prime tower is Kamp has now be incorporated in to main stream klipper many more machines use adaptive bed meshing now and this requires the objects to be identified as a exclude able object. 
 
 Orca has a better way to support adaptive mesh, which is more safe. 
 
 https://github.com/SoftFever/OrcaSlicer/wiki/adaptive-bed-mesh

lijgame, 2025-09-03:
> While I do like this option but it doesn't work
>
>  When your start gcode is all wrapped into a macro like we have to do in our tool changers
>
>
>
>
>
>
>
>
>
>
>
>  lijgame  left a comment [...]

 
The Orca adaptive mesh works exactly that way; it allows you to call the function.  
If you use a single print_start macro in the machine start g-code, just pass these parameters to that macro—there's no difference.

https://github.com/SoftFever/OrcaSlicer/wiki/adaptive-bed-mesh#klipper