OrcaSlicer/OrcaSlicer #7785: Creality k1c 0.2 nozzle profile

Dedoque, 2024-12-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

there's no profile for that printer with a 0.2 nozzle

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Please add 0.2  nozzle printer and filament profiles for Crality k1c

### Describe alternatives you've considered

Modifed other profiles not working as expected

thelegendtubaguy, 2026-11-08:
Hey there!  It looks like Creality doesn't ship .2 profiles for any K1 family printer, so there's nothing for Orca to port over.

You can create your own custom profiles for the .2 nozzles however by setting lower layer height, line width, and speeds.

Since there's nothing for Orca to do here, I suggest closing.

raistlin7447, 2026-11-08:
Thanks for the request. Creality doesn't ship a 0.2 nozzle profile for the K1C, or for any other K1 printer, in CrealityPrint. Orca's K1 profiles come from Creality's, so there's nothing for us to bring over right now. I'm closing this as not planned.

In the meantime you can make your own profile. Start from the K1C 0.4 nozzle preset, set the nozzle diameter to 0.2, and lower the layer height, line widths and speeds to match.

If Creality adds a K1 0.2 profile to CrealityPrint, or someone wants to open a PR with tested K1C 0.2 profiles, comment here and we'll reopen.

closed 2026-11-08 by raistlin7447