OrcaSlicer/OrcaSlicer #7788: Add mouse navigation options like Freecad

FabioM3, 2024-12-18:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Different schemes to mouse buttons for "navigaton styles", something like FreeCad or Fusion 360 have.
It's confusing to use different navigation styles.

The problem is that everyone uses different apps to edit in 3D, and is a needed feature to have the same mouse functionality in regards of camera movement.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I think an option in Settings with some options to change the behavior of the mouse is the simplest approach.
And an option to select the speed too.

MxBrnr, 2024-12-19:
Basically a duplicate of https://github.com/SoftFever/OrcaSlicer/issues/7107

onolox, 2024-12-19:
No is not a duplicate, I ask for diferent ways to control the view with mouse, that #7107 asks for a more general change.
I give a solution to the problem.

I-Dude, 2024-12-29:
I just want to add that I have been looking for a way to change the navigation (3d view) bindings since I started using Orca Slicer.

My two cents on navigation: left click for object manipulation, middle click for pan, right click to rotate, and scroll to zoom. Would be nice to have the following togglable: lock Z axis to be vertical on monitor, rotate around object center (or build plate center, or center on mouse click). 

Its funky that 3d view is free, but then the Z axis snaps to vertical when you pan.

FreeCad's solution is probably the easiest, Its just a drop down selection. Or they could just let us go ham by making everything rebindable.

GabrieleCzz, 2025-02-17:
It would be very convenient to have the same 3D rotation like Fusion 360 pressing the shift key to enable rotation. Using Orcaslicer with a macbook the camera style "touchpad" in the version 2.2 is not working (maybe because alt+move is not available on Mac). I would also mention that the 3D navigation should be around the object and not the plate center (see #7107 )