OrcaSlicer/OrcaSlicer #7790: Error logging into Bambu Cloud

cruzer619, 2024-12-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 and 2.3.0 nightly

### Operating System (OS)

macOS

### OS Version

Ventura 13.7.1

### Additional system information

MacbookPro 13"2017, DualCore i5, 16gb ram

### Printer

Ender5 Pro

### How to reproduce

1. delete all previous version of OrcaSlicer.  And folder in Application support location.
2. copy Orcaslicer to Application folder.
3. launch Orcaslicer
4. install network plugin
5. login with account
6. get error from Importing Model windows stating : Download failed; unknown file format.

### Actual results

Not able able to sync user profiles presets.  Have to force quit app and relaunch app but no sync'd user profiles.

### Expected results

able to run Orcaslicer with sync'd user profiles.

### Project file & Debug log uploads

[orcaslicer-error.log](https://github.com/user-attachments/files/18189127/orcaslicer-error.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

I'm able to login with bambu studio app and sync profiles with no errors.

cruzer619, 2024-12-19:
Ok after playing around and looking at more issues logs #675 .  I determined that my error was because of the 3rd party login.  In my case google login.  When I changed to email login.  Error windows doesn't come up.  But now the issue is syncing profiles between Mac and Windows.  nothing I do will sync profiles over.  The only solution I was able to sync on both mac or windows.  Was to make a printer profile in BambuStudio app and then showed up on both windows orcaslicer and mac orcaslicer.  So i'm guessing something didn't get migrated over???  guessing here.

landoncope, 2024-12-21:
I'm also have this issue, on both 2.2.0 and 2.3.0 (f11eb34).

dnasir, 2024-12-22:
I managed to log in by downloading an older version of OrcaSlicer (v2.1.0), logging in there, and then updating to v2.2.0.

MacBook Pro M2 Max, Sequoia 15.1.1

cruzer619, 2024-12-23:
@dnasir will give that a try.. thanks.

cruzer619, 2024-12-23:
ok after playing around for awhile.  version 2.1.0 does work on mac able to login with google login and able to sync user profiles.  But I can't update to 2.2.0 I still get the login error.  Which i'm assuming has to do with 3rd party login as stated in #675.  If i'm using 2.2.0 and use only an email login, able to login with no errors but not able to sync profiles between mac and windows.  Now with 2.1.0 I do get an error after logging into bambu cloud but gives a browser error.  "There is no application set to open URL:  bambustudioopen://"  but still able to use the app with no issues.  while 2.2.0 doesn't give that browser error but gives the app error I stated above and crashes.  then needing to force quit.  I hope that helps solve the issue..  so for now going to use 2.1.1 for now which works..

theosib, 2024-12-31:
I tried the latest 2.3.0, and it still has this problem. It crashes every time I try to login to bambu cloud. ONE time, I did get the login page, and after logging in, it launched Orca again, but then I got this download error.

theosib, 2025-01-01:
This seems like a pretty serious bug. Do you think it would be a good idea to have a developer mark it as a blocker for the next release?

recrudesce, 2025-01-15:
I have this same issue on my Macbook with latest Orca versions.

Daryl11, 2025-02-28:
I have the same issue with 2.3.0 on Mac Intel.  Also works fine with Bambu Studio.

timbit123, 2025-04-03:
This is an old issue and still no solution? I have the same issue on mac os but Windows is fine

eadium, 2025-04-06:
Encountered this bug today, ver 2.3.0, MacBook Pro 2019 (Intel), macOS Sonoma 14.7, using Google auth. 
Slicer window freezes with loading cursor and showing a pop-up with title "Importing Model" saying that "Download failed; unknown file format." even if I haven't opened anything. Can only force quit the app after this. After restart I can see that I am logged in but no profiles were synced.

luis-godinez, 2025-04-15:
Same issue after updating to 2.3.0

MacOS Sequoia 15.3, M1 Max

adigandhi, 2025-05-04:
Sam issue on Apple M1 Max, when updating to v2.3.0

caiorg, 2025-05-19:
I can confirm that's also my case.
Installed v2.3.0 stable on an Apple M1 Macbook Pro and got the error.
Removing v2.3.0, installing v2.1.1, syncing and only then updating to v2.3.0, solved the issue.

gabrsar, 2025-05-21:
Same issue here. Tried downgrading to v2.2.0,  same issue.

caiorg, 2025-05-21:
Just wanted to point out that I had no issues on my Windows machine.

@gabrsar , I did went straight to v2.1.1, as @cruzer619 mentioned earlier, and was able to solve/workaround the issue.

caiorg, 2025-05-21:
I had some custom filament profiles and they did sync. 

Will do some more testing later today to confirm everything works as expected (after the workaround). 

> > I can confirm that's also my case. Installed v2.3.0 stable on an Apple M1 Macbook Pro and got the error. Removing v2.3.0, installing v2.1.1, syncing and only then updating to v2.3.0, solved the issue.
> 
> @caiorg so your able to sync settings now after you upgraded to 2.3.0 release version??  So if you make a change in windows version and you login into your mac those changes are there?

cruzer619, 2025-05-21:
> I had some custom filament profiles and they did sync.
>
> Will do some more testing later today to confirm everything works as expected (after the workaround).

ok thanks...  keep me posted.

caiorg, 2025-05-22:
@cruzer619 I've created a custom filament on my Windows machine and spun up Orca on my Mac and it was synced.

cruzer619, 2025-05-22:
> [@cruzer619](https://github.com/cruzer619) I've created a custom filament on my Windows machine and spun up Orca on my Mac and it was synced.

I did the same and is working sweet.. have latest release version running 2.3.0.  But if you do a clean install ( remove all versions and remove cache )  And do a vanilla install of 2.3.0 doesn't sync.  So workaround for now.. is install 2.1.x version.. login into cloud.  Then you can upgrade to latest version.  So still broken in latest version ( vanilla install ).  Will now test dev version and see if it sync's.. Thanks again @caiorg for that .

caiorg, 2025-05-22:
> > [@cruzer619](https://github.com/cruzer619) I've created a custom filament on my Windows machine and spun up Orca on my Mac and it was synced.
> 
> I did the same and is working sweet.. have latest release version running 2.3.0.  But if you do a clean install ( remove all versions and remove cache )  And do a vanilla install of 2.3.0 doesn't sync.  So workaround for now.. is install 2.1.x version.. login into cloud.  Then you can upgrade to latest version.  So still broken in latest version ( vanilla install ).  Will now test dev version and see if it sync's.. Thanks again @caiorg for that .
> 
> works on dev version as well if you upgrade from 2.1.X

No worries... Glad to help.
BTW... Can confirm that a clean 2.3.0 breaks cloud sync.

cruzer619, 2025-08-21:
still have issue with vanilla install.

closed 2026-10-30 by github-actions[bot]