OrcaSlicer/OrcaSlicer #7791: Faces in the Lay on Face option do not update when reloading an object from disk

Naautilus, 2024-12-18:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

AMD Ryzen 5 5600x, RTX 3070 Ti

### Printer

Bambu Lap P1S

### How to reproduce

1. Drag an STL file into a new, empty project
2. Press "Lay on Face"
2. Replace the STL file with a different file in the file browser
3. Press "Reload from disk" in the right-click menu of the object
4. Success ! ! !  The next time "Lay on Face" is pressed, the faces will be incorrect.

### Actual results

[image]

When reloading an STL file from disk with a different shape, the faces in the Lay on Face menu do not update. The faces that are present are still functional and rotate the mesh properly.

### Expected results

The faces are updated.

### Project file & Debug log uploads

[my_bugged_file.zip](https://github.com/user-attachments/files/18189183/my_bugged_file.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

discip, 2024-12-19:
Unfortunately this is also present in the latest dev build.

Noisyfox, 2025-01-14:
Same as https://github.com/prusa3d/PrusaSlicer/issues/5127

PR #8036 (2025-01-22): Fix issue that flatten gizmo faces not updated after reload from disk
Fix #7791
closed 2025-01-22 by Noisyfox