OrcaSlicer/OrcaSlicer #7792: Z hop when travel

DrumClock, 2024-12-19:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Hi,
I have been printing for a long time without using "Z hop when retracting" and there are no problems.
I would like a new option to configure when the tool head is traveling.

Add the option "Z hop when travel"

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Z hop při cestovní rychlosti (Travel speed)

[image]

fritzw, 2024-12-19:
What would be the difference to the current option "Z hop when retracting"? Retraction is only performed before travel, so unless I misunderstood your intention, it is essentially the same as "Z hop when travel", right?

Or do you want to disable it? When you disable "Z hop when retracting", the slicer should include no more Z hops, so if you still see Z hops in the gcode, that would be strange.

DrumClock, 2024-12-19:
Hi,
yes, I expressed myself incorrectly. I meant that the Z hop
is only performed when the part is printed and the next part is traveled.

For example:
- the infill print is completed
- the z hop is performed
- the toolhead moves to the next print position (travel speed)
- the z hop returns
- the next part perimeter print starts

'Z hop when retraction' works correctly, there is no problem here.

This would limit the z hop to a minimum (and print time) and would not have to be performed at every G10/G11, so also in the infill and perimeters of the part.

I hope I have put it in perspective now.