OrcaSlicer/OrcaSlicer #7795: Pressure Advance Pattern Test Using Wrong Layer Height on Different Nozzle Sizes

jryan15, 2024-12-19:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

windows 11

### Additional system information

dell laptop, ryzen processor, amd graphics

### Printer

Bambu P1S

### How to reproduce

Use a 0.2mm nozzle (or other non-0.4mm size)
Click Calibration up the top.
Select "Pressure Advance".
Select "PA Pattern" and hit OK.

### Actual results

The automated test generation tries to give us a layer height of 0.2mm (with 0.25mm for the initial layer), resulting in a slicing error because the layer height can't exceed the nozzle diameter.  (same as issue #6672 https://github.com/SoftFever/OrcaSlicer/issues/6672)

[image]

### Expected results

It should adjust the layer height according to the nozzle diameter. It does fine for the line width.

Also not sure if changing the layer height manually messes up the PA test. That might be a separate bug.

### Project file & Debug log uploads

[Pressure Advance Test_BAD_0.2mm.3mf.zip](https://github.com/user-attachments/files/18200324/Pressure.Advance.Test_BAD_0.2mm.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

Not sure why there was no follow up on the previous bug report.

MxBrnr, 2024-12-19:
Those are just default settings. You are expected to change these manually if you want/need them to be different.
https://github.com/SoftFever/OrcaSlicer/wiki/Calibration#pattern-method
[image]

jryan15, 2024-12-20:
Thank you!  I was able to modify it and calibrate for the smaller nozzle.

FWIW I would have tried this, but noticed it generated lots of G-code which made me think re-adjusting the layer heights was not so simple.  If the pattern generator is agnostic to layer height and other settings, it would probably be much less confusing if it did not override them to begin with.

MxBrnr, 2024-12-22:
I agree, and would like to see this change.

closed 2025-06-09 by github-actions[bot]