OrcaSlicer/OrcaSlicer #7799: More seam orientatios or rotate seam witch object

Zogar89, 2024-12-20:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When optimizing bed space, some objects may be better printed interlaced, which means that one object should have the seam on the back and another on the front. But this last option does not exist.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add more orientations for the seam apart from the rear option. 

[image]

### Describe alternatives you've considered

In Cura you can select 8 positions and even select the position in bed coordinates to which all the seams will point. The option becomes really interesting when applied to each object, allowing you to interlace objects with seams in front and behind, optimizing space, or left and right.

soundtechscott, 2025-01-01:
you can do this in per object settings... select where you want the seam or draw it on manually

[image]

I just drew some in random places for show and tell

[image]

Zogar89, 2025-01-02:
@soundtechscott NO.

I don't want to paint the models manually in order to have a fixed seam.
Painted seams are awkward to do when you have a lot of models and are not very neat since they tend to have more zig zag than automatic ones.

It is easy to add more orientations than just the back.