OrcaSlicer/OrcaSlicer #7800: prime tower change size when adding filaments

RaySajuuk, 2024-12-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

15.2 (24C101)

### Additional system information

M3 Max

### Printer

Bambu Lab X1C

### How to reproduce

1. add any model e.g. Orca Cube. Or use provided 3mf file.
2. add second filament.
3. change any part of the model or paint use second filament.
4. make sure prime tower is enabled and showing on the build plate.
5. add multiple filaments.
6. observe prime tower changes.

### Actual results

prime tower will change size on Y-axis.
[image]

### Expected results

prime tower should not be affected by adding filaments
[image]

### Project file & Debug log uploads

[OrcaCube_PrimeTower_Test.3mf.zip](https://github.com/user-attachments/files/18206384/OrcaCube_PrimeTower_Test.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

RaySajuuk, 2024-12-20:
Issue persists and exaggerated in current dev build with commit 27ec198.

[image]

RaySajuuk, 2024-12-20:
Additionally verified workaround by toggle Precision Z height does not work.

RaySajuuk, 2024-12-31:
I seem to have found a workaround.
I will slice the plate, the issue resolves after slicing. I then go back to prepare and arrange the prime tower.
Use either auto arrange all or current plate only will cause the issue to reappear.

Ohmarinus, 2025-01-24:
> I seem to have found a workaround. I will slice the plate, the issue resolves after slicing. I then go back to prepare and arrange the prime tower. Use either auto arrange all or current plate only will cause the issue to reappear.

Thanks for mentioning. This helped me get through as well. Slicing and going back to 'Prepare' turned the purge tower back to its normal size. I could then move the purge tower to a more convenient spot without it jumping back into its original confusing size. Slicing again then also reveals a normal purge tower.

[image]

Printing right now and all seems to go as planned.

closed 2025-05-03 by github-actions[bot]