OrcaSlicer/OrcaSlicer #7803: Pop!_Os Support

CodKid1710, 2024-12-20:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I'm annoyed when building Orca Slicer on Pop!_Os, (which is a varient of Ubuntu) since is says that 
`Your distribution does not appear to be currently supported by these build scripts`

 

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

It should automatically detect these variants of linux
If that is TOO difficult it would be nice to be able to automatically override this error

Also you could use ID_LIKE to see whether it is a varient of ubuntu

### Describe alternatives you've considered

1. To fix this i simply added ` || [ "${DISTRIBUTION}" == "pop" ]` to the end of line of 83\
2. You could also add something like this `|| [  "${DISTRIBUTION_LIKE}" == "ubuntu debian" ] `

cochcoder, 2025-01-16:
Does the issue go away if you replace line 83 with `if [ "${DISTRIBUTION_LIKE}" == "ubuntu" ] || [ "${DISTRIBUTION_LIKE}" == "debian" ]`? 

I ask as replacing the current values with `[  "${DISTRIBUTION_LIKE}" == "ubuntu debian" ]` break GitHub actions.

CodKid1710, 2025-01-17:
I'll get back to you in a few days, i am currently on vacation, thanks

CodKid1710, 2025-01-17:
I managed to fix it change this

```shell
DISTRIBUTION=$(awk -F= '/^ID=/ {print $2}' /etc/os-release)
# treat ubuntu as debian
if [ "${DISTRIBUTION}" == "ubuntu" ] || [ "${DISTRIBUTION}" == "linuxmint" ]
```

To this
```shell
DISTRIBUTION_LIKE=$(awk -F= '/^ID_LIKE=/ {print $2}' /etc/os-release)
# treat any ubuntu based os as debian
if [ "${DISTRIBUTION_LIKE}" == "\"ubuntu debian\"" ]
``` 
It should allow for ANY debian or ubuntu based os to compile it

grooby, 2025-02-13:
Another issue I encountered trying to build this in Pop_OS was CMake not able to find Ninja.  I had to install ninja-build in order to compile

cochcoder, 2025-02-28:
@CodKid1710 

I've had to change a few things to make it work for Github Actions, if you change lines 81-89 to the code below, does it still work for you? 

```
DISTRIBUTION=$(awk -F= '/^ID=/ {print $2}' /etc/os-release | tr -d '"')
DISTRIBUTION_LIKE=$(awk -F= '/^ID_LIKE=/ {print $2}' /etc/os-release | tr -d '"')
# Check for direct distribution match to Ubuntu/Debian
if [ "${DISTRIBUTION}" == "ubuntu" ] || [ "${DISTRIBUTION}" == "linuxmint" ]; then
    DISTRIBUTION="debian"
# Check if distribution is Debian/Ubuntu-like based on ID_LIKE
elif [[ "${DISTRIBUTION_LIKE}" == *"debian"* ]] || [[ "${DISTRIBUTION_LIKE}" == *"ubuntu"* ]]; then
    DISTRIBUTION="debian"
fi
```

CodKid1710, 2025-02-28:
Yeah it works! 

Thanks

CodKid1710, 2025-02-28:
should i close this issue?

cochcoder, 2025-03-01:
>  Yeah it works!
> 
> Thanks

Great! 

> should i close this issue?

You don't need to, once the PR is merged this issue will be closed automatically.

PR #8625 (2025-03-01): Allow compilation on distributions based on Ubuntu/Debian
Fixes #7803

If a Linux distribution is based on Ubuntu/Debian, it will compile Orca Slicer rather than giving a unsupported distro error.
closed 2025-03-01 by Noisyfox