OrcaSlicer/OrcaSlicer #7804: Underextrusion after overhangs because smoothing flow/speed is not working

d-schmidt, 2024-12-20:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Creality K1C

### How to reproduce

Enable fast speeds and smoothing and have slow overhangs

### Actual results

Underextrusion:
[image]

### Expected results

It takes a moment to accelerate, which seems to work at some places but these corners cause it to switch to full speed immediatly

### Project file & Debug log uploads

[worx battery holder-orange.zip](https://github.com/user-attachments/files/18215678/worx.battery.holder-orange.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

d-schmidt, 2024-12-20:
It looks like the layer time limit is exacerbating the problem. It moves faster on layers where it has to slow down for overhangs.
The speed itself isn't a problem, the printer and filament can keep up. given enough time to get the flow up again.

[image]

d-schmidt, 2024-12-20:
Also somethings seems funky with the Smoothing segment length:

1:
[image]

2:
[image]

3:
[image]

4:
[image]

5:
[image]

Otarifpv, 2024-12-27:
I'm also suffering from this problem, I've been trying to deal with the smoothing, but I can only hide it, I can't get it to print correctly, I also see that there are layers in which the printer for some reason triggers the layer time, and cooling is not everything as regular as I would like, no matter how I configure it I never get what I expected, I also think that some bridges fail for this same reason, the area where it is anchored on the perimeters used a speed that was too high and this would cause the bridge not to anchor correctly.

excuse my english

d-schmidt, 2025-01-15:
Nobody got any idea how smoothing, layer time and overhang mess each other up?

Also for asthetic reasons the layertime shouldn't switch this hard. A 180mm/s surface will never look the same as the 80mm/s layer below.

fridaykillme, 2025-01-27:
Im not sure what i have same problem, but very similar. FF ad5m pro. Early i have not underextrusion after overhangs/turns.

igiannakas, 2025-01-31:
apply extrusion rate smoothing - latest nightly also has an option to limit this to external perimeters only.

d-schmidt, 2025-01-31:
> apply extrusion rate smoothing - latest nightly also has an option to limit this to external perimeters only.

@igiannakas the last screenshots are with extrusion rate smoothing. It is bugged. I added a screenshot for each (1-5) extrusion rate smoothing length

igiannakas, 2025-01-31:
Set segment length to 1, ERS speed to 50 and only on outer walls.  Segment size is how big the length of the step in mm where the slowdown happens. ERS speed is the rate of volumetric change that is allowed.

[image]

You can go below 1 if needed, which will give you a smoother transition at the risk of stuttering if the printer's control board cannot keep up with the segment size. Eg below is 0.5mm length.

[image]

You shouldn't need to use it through. What you are seeing is the extruder struggling to keep up with speed changes. Is your PA tuned accurately?

igiannakas, 2025-01-31:
> Nobody got any idea how smoothing, layer time and overhang mess each other up?
> 
> Also for asthetic reasons the layertime shouldn't switch this hard. A 180mm/s surface will never look the same as the 80mm/s layer below.

I tried introducing layer time smoothing but the effects were marginally worse actually.https://github.com/SoftFever/OrcaSlicer/pull/8107  The sharp transition was spread over multiple layers. Suggestion is to keep constant speeds on outer walls which helps a ton with model surface appearance. this can be found in the filament settings - dont slow down outer walls.

d-schmidt, 2025-01-31:
The nightly build still has these "artifacts" throwing off my printer.

The default 300 - 3 looks like this:
[image]
These can't be working as intended. Given a bit time the printer handles everything fine. But these hard magnitude speed changes are too much. 

I will test wether my printer can handle the length of 1 mm. The stable has a warning there. So far this seemse the only length working well. At least in the preview.

igiannakas, 2025-01-31:
Did you try the settings I set above? 50 ERS and 1mm segment? The screenshots were taken with the project file and these two settings set as per the screenshot. 300 ERS may be too high for your specific speed settings.

If you want attach your latest project file, but first please try the settings I suggested.

The build I'm using is here: https://github.com/igiannakas/OrcaSlicer/actions/runs/13077675455 

It will be done compiling in around 1h. Its a few commits ahead of the nightly.

d-schmidt, 2025-01-31:
I did try 50 ERS and 1 mm. It looks much better and I will check wether my pritner can handle the extra load. It's an K1C so I hope it works.

The screenshots are from the "default" 300-3 settings.

50-1 looks like this for mee in the latest available nigthly build:

[image]

igiannakas, 2025-01-31:
Looks about right. I presume the project file you've got open right now is slightly different to the one in the start of the thread? In any case, yes these will be more gradual.

However, the printer really shouldn't need this setting - it's there to remove minor surface imperfections, but what you're getting is major under extrusion. Its supposed to correct issues like this

[image] https://github.com/SoftFever/OrcaSlicer/pull/2161

Which are not as severe as the ones you're seeing. What you are seeing is caused by incorrect PA and / or an extruder issue and / or excessive drag in the filament path.

d-schmidt, 2025-01-31:
Okay, I will try to calibrate PA for that fillament.

I just assumed it is something smoothing should help with as the print only failed at the 10->110mm/s switch at that place because the slicer wants to make up for the lost time printing the overhang. If PA should actually prevent these artifacts after magnitude speed-switches I will put some time into them.

igiannakas, 2025-01-31:
Correct. Pa is a static value set per filament. There’s nothing to see in the preview as it’s directly sent to the printer to use in its extruder planning routines.

PA can make absolutely a huge difference in speed transitions. That’s what it’s made for :)

Suggest to have a read here: https://ellis3dp.com/Print-Tuning-Guide/articles/pressure_linear_advance/introduction.html

d-schmidt, 2025-01-31:
You want me to keep this ticket open for the weird smoothing behavior or should I close it?

[image]

hotellonely, 2025-02-05:
Many people are having the similar issue but it looks like we don't have a definitive answer.

1. PA
PA, at least a static PA, won't be able to solve this issue. Many have tried including myself. A dynamic PA calibration for multi material or various material is tooooo time costly.

2. Extruder
Even very popular fast printing machiens like X1C, K1C are having this issue. 

3. ESR/Smooth speed
**No, even ESR cannot fully fix it. Even with a very low extrusion smoothing threshold, the under extrusion is still showing up.**

4. Turning off Slow Down for Overhang
It DOES help! But at the cost of overhang performance. 

5. Misplaced seams
Sometimes slicer would create seams on overhang area. Causing the overextrusion to layer up. Eventually causing a micro-clog.

6. **WARP**, or **environment too hot**, or not enough minimum layer time.
Quite important factor actually. Small overhangs like these tend to warp up. Part cooling in this case don't help much, you need the environment to be cool. At least that is how I mitigated much of my exactly the same problem.

I don't know the exact reason, but I've explored many options, so here's my two cent.

closed 2025-05-14 by github-actions[bot]

sebastianha, 2025-11-08:
I was pretty sure I already did this but I did it again for this filament and I needed to increase it a lot. I think now it's too much but definitely thanks for the hint!