OrcaSlicer/OrcaSlicer #7820: Thick Bridge Min. Length Threshold

MxBrnr, 2024-12-22:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

When using `Thick bridge` settings, add a setting for `Maximum thin bridge length threshold` (or similarly worded).

### Additional context

This will allow the use of thin bridges for shorter distances (less than the set threshold) to provide better aesthetics where the machine is capable, and thick bridges for longer distances (greater than set threshold) where the reliability of the bridge line is required. This feature could also be combined with being able to specify individual print speeds for short vs long bridges where more cooling time may be needed.

CCS86, 2025-02-13:
Seems like a good idea.

stur0067, 2025-07-04:
I like this idea