OrcaSlicer/OrcaSlicer #7828: Frequent slicing crashes using EndeavourOS

dwdbmeagle, 2024-12-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

Kernel Version: 6.12.4-zen1-1-zen (64-bit)

### Additional system information

system information:

> Operating System: EndeavourOS 
KDE Plasma Version: 6.2.4
KDE Frameworks Version: 6.9.0
Qt Version: 6.8.1
Kernel Version: 6.12.4-zen1-1-zen (64-bit)
Graphics Platform: Wayland
Processors: 6 × AMD Ryzen 5 3500X 6-Core Processor
Memory: 15.5 GiB of RAM
Graphics Processor: AMD Radeon RX 6700
Manufacturer: Micro-Star International Co., Ltd.
Product Name: MS-7C91
System Version: 1.0

locale information:

> locale
LANG=en_US.UTF-8
LC_CTYPE=en_US.UTF-8
LC_NUMERIC=en_GB.UTF-8
LC_TIME=en_GB.UTF-8
LC_COLLATE="en_US.UTF-8"
LC_MONETARY=en_DK.UTF-8
LC_MESSAGES="en_US.UTF-8"
LC_PAPER=en_GB.UTF-8
LC_NAME=en_GB.UTF-8
LC_ADDRESS=en_DK.UTF-8
LC_TELEPHONE=en_GB.UTF-8
LC_MEASUREMENT=en_DK.UTF-8
LC_IDENTIFICATION=en_DK.UTF-8
LC_ALL=

### Printer

V400

### How to reproduce

Slice a slightly complicated model

### Actual results

orca-slicer just crashes

strace result of crash:
> /usr/include/c++/14.2.1/bits/stl_vector.h:1130: std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Slic3r::PerimeterGeneratorArachneExtrusion; _Alloc = std::allocator<Slic3r::PerimeterGeneratorArachneExtrusion>; reference = Slic3r::PerimeterGeneratorArachneExtrusion&; size_type = long unsigned int]: Assertion '__n < this->size()' failed.
[{WIFSIGNALED(s) && WTERMSIG(s) == SIGABRT && WCOREDUMP(s)}], 0, NULL) = 50703
rt_sigaction(SIGINT, {sa_handler=SIG_DFL, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x735f6f5fd1d0}, {sa_handler=0x6031a52c61e0, sa_mask=[], sa_flags=SA_RESTORER, sa_restorer=0x735f6f5fd1d0}, 8) = 0
ioctl(2, TIOCGWINSZ, {ws_row=74, ws_col=115, ws_xpixel=923, ws_ypixel=1111}) = 0
openat(AT_FDCWD, "/usr/share/locale/locale.alias", O_RDONLY|O_CLOEXEC) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=2998, ...}) = 0
read(3, "# Locale name alias data base.\n#"..., 4096) = 2998
read(3, "", 4096)                       = 0
close(3)                                = 0
openat(AT_FDCWD, "/usr/share/locale/en_ZA/LC_MESSAGES/bash.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en/LC_MESSAGES/bash.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en_GB/LC_MESSAGES/bash.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
fstat(2, {st_mode=S_IFCHR|0600, st_rdev=makedev(0x88, 0), ...}) = 0
openat(AT_FDCWD, "/usr/share/locale/en_ZA/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en/LC_MESSAGES/libc.mo", O_RDONLY) = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/usr/share/locale/en_GB/LC_MESSAGES/libc.mo", O_RDONLY) = 3
fstat(3, {st_mode=S_IFREG|0644, st_size=1430, ...}) = 0
mmap(NULL, 1430, PROT_READ, MAP_PRIVATE, 3, 0) = 0x735f6f840000
close(3)                                = 0
write(2, "/usr/bin/orca-slicer: line 5: 50"..., 86/usr/bin/orca-slicer: line 5: 50703 Aborted                 (core dumped) $0-bin "$@"
) = 86
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
--- SIGCHLD {si_signo=SIGCHLD, si_code=CLD_KILLED, si_pid=50703, si_uid=1000, si_status=SIGABRT, si_utime=4761 /* 47.61 s */, si_stime=376 /* 3.76 s */} ---
wait4(-1, 0x7fff5e50bed0, WNOHANG, NULL) = -1 ECHILD (No child processes)
rt_sigreturn({mask=[]})                 = 0
read(255, "", 104)                      = 0
rt_sigprocmask(SIG_BLOCK, [CHLD], [], 8) = 0
rt_sigprocmask(SIG_SETMASK, [], NULL, 8) = 0
exit_group(134)                         = ?
+++ exited with 134 +++

### Expected results

Slice and not crash

### Project file & Debug log uploads

[debug_Mon_Dec_23_18_06_26_53994.log.0.zip](https://github.com/user-attachments/files/18231737/debug_Mon_Dec_23_18_06_26_53994.log.0.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

dwdbmeagle, 2024-12-24:
After some further testing.
This only occurs with the arch provided package. AppImage version does not seem to have this problem. Slices the same model without a problem.

flaviut, 2025-01-06:
I can reliably reproduce this. It's only an issue with Arachne, and for this particular model, only when 3 walls are used:

[image]

[motor mount.3mf.zip](https://github.com/user-attachments/files/18325142/motor.mount.3mf.zip)

Stack trace:

```
Thread 59 "slic3r_tbb_20" received signal SIGABRT, Aborted.
[Switching to Thread 0x7fff41bf66c0 (LWP 2530462)]
__pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
44	     return INTERNAL_SYSCALL_ERROR_P (ret) ? INTERNAL_SYSCALL_ERRNO (ret) : 0;
(gdb) bt
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007fffefaa5463 in __pthread_kill_internal (threadid=<optimized out>, signo=6) at pthread_kill.c:78
#2  0x00007fffefa4c120 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007fffefa334c3 in __GI_abort () at abort.c:79
#4  0x00007ffff7cd3af0 in std::__glibcxx_assert_fail
    (file=file@entry=0x55555a499538 "/usr/include/c++/14.2.1/bits/stl_vector.h", line=line@entry=1130, function=function@entry=0x55555a5204d0 "std::vector<_Tp, _Alloc>::reference std::vector<_Tp, _Alloc>::operator[](size_type) [with _Tp = Slic3r::PerimeterGeneratorArachneExtrusion; _Alloc = std::allocator<Slic3r::PerimeterGeneratorArachneExt"..., condition=condition@entry=0x55555a40d87f "__n < this->size()") at /usr/src/debug/gcc/gcc/libstdc++-v3/src/c++11/assert_fail.cc:41
#5  0x0000555556b4a5f1 in std::vector<Slic3r::PerimeterGeneratorArachneExtrusion, std::allocator<Slic3r::PerimeterGeneratorArachneExtrusion> >::operator[]
    (this=<optimized out>, __n=<optimized out>) at /usr/include/c++/14.2.1/bits/stl_vector.h:1130
#6  Slic3r::PerimeterGenerator::process_arachne (this=this@entry=0x7fff41be8180)
    at /usr/src/debug/orca-slicer/OrcaSlicer/src/libslic3r/PerimeterGenerator.cpp:3177
#7  0x0000555556ae7d41 in Slic3r::LayerRegion::make_perimeters
    (this=this@entry=0x7ffc78015cc0, slices=..., compatible_regions=std::vector of length 1, capacity 1 = {...}, fill_surfaces=fill_surfaces@entry=0x7ffc78015d38, fill_no_overlap=fill_no_overlap@entry=0x7ffc78015d50) at /usr/src/debug/orca-slicer/OrcaSlicer/src/libslic3r/LayerRegion.cpp:116
#8  0x0000555556ae2866 in Slic3r::Layer::make_perimeters (this=0x7ffc7800c1e0) at /usr/src/debug/orca-slicer/OrcaSlicer/src/libslic3r/Layer.cpp:207
#9  0x00005555566da0c6 in operator() (__closure=<optimized out>, range=<optimized out>) at /usr/src/debug/orca-slicer/OrcaSlicer/src/libslic3r/PrintObject.cpp:389
#10 tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<long unsigned int>, Slic3r::PrintObject::make_perimeters()::<lambda(const tbb::detail::d1::blocked_range<long unsigned int>&)>, const tbb::detail::d1::auto_partitioner>::run_body (this=<optimized out>, r=<optimized out>)
    at /usr/src/debug/orca-slicer/dep_linux/usr/local/include/oneapi/tbb/parallel_for.h:119
#11 tbb::detail::d1::dynamic_grainsize_mode<tbb::detail::d1::adaptive_mode<tbb::detail::d1::auto_partition_type> >::work_balance<tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<long unsigned int>, Slic3r::PrintObject::make_perimeters()::<lambda(const tbb::detail::d1::blocked_range<long unsigned int>&)>, const tbb::detail::d1::auto_partitioner>, tbb::detail::d1::blocked_range<long unsigned int> > (this=0x7fffc4383468, start=warning: RTTI symbol not found for class 'tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<unsigned long>, Slic3r::PrintObject::make_perimeters()::{lambda(tbb::detail::d1::blocked_range<unsigned long> const&)#2}, tbb::detail::d1::auto_partitioner const>'
..., range=..., ed=...)
    at /usr/src/debug/orca-slicer/dep_linux/usr/local/include/oneapi/tbb/partitioner.h:462
#12 tbb::detail::d1::partition_type_base<tbb::detail::d1::auto_partition_type>::execute<tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<long unsigned int>, Slic3r::PrintObject::make_perimeters()::<lambda(const tbb::detail::d1::blocked_range<long unsigned int>&)>, const tbb::detail::d1::auto_partitioner>, tbb::detail::d1::blocked_range<long unsigned int> > (this=0x7fffc4383468, start=warning: RTTI symbol not found for class 'tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<unsigned long>, Slic3r::PrintObject::make_perimeters()::{lambda(tbb::detail::d1::blocked_range<unsigned long> const&)#2}, tbb::detail::d1::auto_partitioner const>'
..., range=..., ed=...)
    at /usr/src/debug/orca-slicer/dep_linux/usr/local/include/oneapi/tbb/partitioner.h:288
#13 tbb::detail::d1::start_for<tbb::detail::d1::blocked_range<long unsigned int>, Slic3r::PrintObject::make_perimeters()::<lambda(const tbb::detail::d1::blocked_range<long unsigned int>&)>, const tbb::detail::d1::auto_partitioner>::execute(tbb::detail::d1::execution_data &) (this=0x7fffc4383400, ed=...)
    at /usr/src/debug/orca-slicer/dep_linux/usr/local/include/oneapi/tbb/parallel_for.h:172
#14 0x0000555556ca7ba7 in tbb::detail::r1::task_dispatcher::local_wait_for_all<false, tbb::detail::r1::outermost_worker_waiter>
    (t=0x7fffc4383400, this=0x7fffd96ecd00, waiter=...)
    at /usr/src/debug/orca-slicer/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/src/tbb/task_dispatcher.h:322
#15 tbb::detail::r1::task_dispatcher::local_wait_for_all<tbb::detail::r1::outermost_worker_waiter> (t=0x0, this=0x7fffd96ecd00, waiter=...)
    at /usr/src/debug/orca-slicer/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/src/tbb/task_dispatcher.h:463
[... cut ...]
    at /usr/src/debug/orca-slicer/OrcaSlicer/deps/build/dep_TBB-prefix/src/dep_TBB/src/tbb/private_server.cpp:221
#20 0x00007fffefaa339d in start_thread (arg=<optimized out>) at pthread_create.c:447
#21 0x00007fffefb2849c in __GI___clone3 () at ../sysdeps/unix/sysv/linux/x86_64/clone3.S:78
```

This is using the https://aur.archlinux.org/packages/orca-slicer package. This package has a couple very minimal patches, but they are unrelated and just there to allow the project to build. Version 2.2.0 from git.

dwdbmeagle, 2025-01-07:
My default is Arachne with 3 wall loops.

closed 2025-04-15 by github-actions[bot]