OrcaSlicer/OrcaSlicer #7830: Scarf joints will not generate with any settings

markaudacity, 2024-12-23:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Win11

### Printer

Bambu X1C

### How to reproduce

Enable scarf joint seams

### Actual results

No scarf seam is generated, layer heights remain consistent and traditional seams are generated

### Expected results

Scarf joint seams should be created

### Project file & Debug log uploads

[debug_Mon_Dec_23_12_29_04_27728.log.zip](https://github.com/user-attachments/files/18232806/debug_Mon_Dec_23_12_29_04_27728.log.zip)
[rM2 Marker inkwell.zip](https://github.com/user-attachments/files/18232808/rM2.Marker.inkwell.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Ergonomicmike, 2024-12-24:
I thought that I had read somewhere that enabling some of the seam settings in Process and/or Retraction settings in Printer would prevent scarf seam. But I can't find it now. 

If you haven't already read [this article from Printables, start here](https://www.printables.com/model/783313-better-seams-an-orca-slicer-guide-to-using-scarf-s)?

cadriel, 2025-01-14:
I'm having this issue too. 8mm cyclinders printed up right - no combination of seam settings works. I've tried increasing the size to 30mm without success.

Also with a bambu x1c, no filament setting / override i can see SHOULD cause this?

Is this a known issue?

cadriel, 2025-01-14:
Perhaps this is related to;
https://github.com/SoftFever/OrcaSlicer/issues/7326

After investigating my slicer, flow would indicate the seam settings are applying - but the print results in a visible seam - much like the above issue.

markaudacity, 2025-01-29:
> I learned that a scarf seam isn't always best.

That may be so, but it's also not never best, and the feature should either function or be removed. 

My use case (engineering test parts) doesn't care about appearance, but it does care about dimensional accuracy, and the ability to "spread" the seam like this instead of having it in the [can't-fully-tune-it-out-no-matter-how-Pepe-Silvia-you-get-about-it] blob of a traditional seam would be a huge get for me. Currently I'm having to modify cylindrical bosses and bores with grooves to hide the seams in, which kinda sucks.

cadriel, 2025-01-29:
fwiw - if I print with the same settings in Bambu Lab, the seams work fine. Pretty sure this is just a bug as linked above. :)

closed 2025-05-07 by github-actions[bot]