OrcaSlicer/OrcaSlicer #7834: Insert Pause creates pause at layer+1

jctune, 2024-12-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0-devel

### Operating System (OS)

macOS

### OS Version

Sequoia 15.2

### Printer

X1C

### How to reproduce

1.  Slice a plate that contains an object that needs a pause
2.  In the preview window, go to the layer that needs a pause.
3.  Slice the plate again
[image]

[image]

### Actual results

Pause gets inserted at the beginning of the next layer.  While the slider on the right will indicate that the pause is on the correct layer, the next layer will be grey in the preview window indicating that it will pause here.  Actual print behavior is to pause at desired layer +1.

[image]

A review of actual exported G-Code for this project also has the pause code inserted on desired layer +1
[image]

### Expected results

The layer where the pause should occur should be both highlighted in grey in the preview window and the printer should actually pause at the beginning of that layer.

### Project file & Debug log uploads

[Gridfinity bin 2x6x5_PLA_4h32m.gcode.zip](https://github.com/user-attachments/files/18241749/Gridfinity.bin.2x6x5_PLA_4h32m.gcode.zip)
[SH2.3mf.zip](https://github.com/user-attachments/files/18241751/SH2.3mf.zip)
[debug_Tue_Dec_24_12_43_23_8452.log.0.zip](https://github.com/user-attachments/files/18241753/debug_Tue_Dec_24_12_43_23_8452.log.0.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Noisyfox, 2025-01-12:
It currently doesn't work well with adaptive layer height.

Noisyfox, 2025-01-12:
It appears to be a rounding error caused by the adaptive layer height.

PR #8017 (2025-01-22): Fix wrong custom gcode layer caused by rounding error
Fix #7834

[image]
closed 2025-01-22 by Noisyfox