OrcaSlicer/OrcaSlicer #7835: Orca deletes all Bambu Studio settings

MULAWI, 2024-12-24:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Prozessor	Intel(R) Core(TM) i5-6400 CPU @ 2.70GHz   2.70 GHz
RAM                16.0 GB
GraficCard       NVIDIA GeForce GTX 970

### Printer

Bambu Lab P1S with AMS

### How to reproduce

1. Open OrcaSlicer
2. Device
3. see what is showing in the slots, in my case is all empty even it was filled in Bambu Studio. After that Bambu is also empty in DEVICE SEction

### Actual results

If I have set everything on the AMS and close Bambu Studio and then start Orca slicer, all the settings are gone and Orca slicer does not recognize anything on the AMS. As if they were fighting each other. 

This was not the case until now, but after updating to 2.2 it is no longer user-friendly.

### Expected results

That they don't fight each other and no matter which slicer I open, the same settings are available.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18241889/log.zip)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Anything else?

[image]
[image]
[image]

wwebers, 2024-12-26:
I had the very same issue with OrcaSlicer 2.2.0 under Ubuntu/Linux. All my AMS settings where cleared. Additionally, I could not change the colors when trying to setting it up again.
However, I realized this issue AFTER changing to LAN mode and then back again. First after setting everything up again in BambuLab even OrcaSlicer worked again.

I suspect they messed up the **LAN mode** somehow. Haven't checked to do a factory reset of everything after changing to LAN mode and start over again as I wouldn't like to loose all the filament and process settings. Shouldn't be nessacary anyway.

wwebers, 2025-02-22:
I'm running X1Plus now, but with the latest AMS firmware. Printer is running completely in LAN mode. I can confirm that only BambuStudio works and keeps the AMS filament configuration. As soon as I connect OrcaSlicer to the printer the configuration is completely messed up, specifically for non-Bambu filaments. Even though I made sure I configured these filaments in OrcaSlicer **before** I configured the AMS slots.

closed 2025-05-31 by github-actions[bot]