OrcaSlicer/OrcaSlicer #7838: Outer wall speed higher than inner.

uki-3D, 2024-12-25:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Win 10

### Printer

Bambu P1S

### How to reproduce

1. Slice file (more complicated one)
2. Select default profile for default printer and filament (eg. 0.20mm Standard @BBL X1C, Bambu Lab P1S 0.4 nozzle, Bambu PETG Basic)
3. Slice file
4. Scroll down on layer preview and check inner/outer wall speed

### Actual results

Outer wall is printed faster than inner wall despite the speed settings.

### Expected results

Outer wall should be printed slower than inner wall.

### Project file & Debug log uploads

[example.zip](https://github.com/user-attachments/files/18247271/example.zip)
[image]
[image]

### Checklist of files to include

- [ ] Log file
- [X] Project file

Noisyfox, 2025-01-12:
[image]
You asked for a faster speed for outer wall than inner wall.

Noisyfox, 2025-01-12:
The speed is limited by the max volumetric speed. Since the inner wall width (0.45mm) is greater than outer wall width (0.42mm), the outer wall speed is faster.

endorphin3d, 2025-01-23:
@Noisyfox thanks so much for this answer. I was going crazy trying to figure out why my outer wall speed wasn't being respected and this ended up being exactly it.

closed 2025-01-23 by uki-3D