OrcaSlicer/OrcaSlicer #7841: Full Contact Support - interface layers to shallow.

BlackNet, 2024-12-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

11

### Additional system information

not hardware related.

### Printer

Voron 2.4, Bambu, all others.

### How to reproduce

Add 1.3mm+ ridges on the bottom of the print for full contact support, like dissolving filament with ASA. 

### Actual results

prints 3 bottom layers of interface, layers of regular support material, 3 top layers of interface material.

### Expected results

clean dissolving interface layers without regular support filament obstructing

### Project file & Debug log uploads

na

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

When using interface layers, there is a need to go beyond 3 bottom and 3 top layers. Deep grooves of 1.2mm+ should get interface layers that are not obstructed by regular support material. When using dissolving filaments, like Aquasys or even PVA, the regular support material obstructs the dissolving filament and hinders removal.

[image]

The squiggly lines should be interface layers; there are three interfaces below and three above them. The squiggly lines are also not attached and free-floating.

Noisyfox, 2025-01-05:
Any 3mf files available for testing?

BlackNet, 2025-01-05:
Yes, here is the file.  Also, it is zipped because GitHub does not support 3MF.
[export - for GitHub Shuttle MGN12 (With belt grooves).zip](https://github.com/user-attachments/files/18310812/export.-.for.github.Shuttle.MGN12.With.belt.grooves.zip)
 files.

Note that layers 42-44 are interface layers; layer 45 is regular filament (non-dissolvable), and 46-48 are dissolvable.

The top and bottom interface layers are set to 3; they should be raised higher in this example. However, we cannot do that, and quite frankly, 3 is just too little in most cases.

BlackNet, 2025-01-05:
Perhaps the best route is let the user choose how many interface layers there will be.

BlackNet, 2025-03-11:
Is there any update on this?  Is it possible to get it in the new release?