OrcaSlicer/OrcaSlicer #7842: Unnecessary preheat of T0 if only T1 is used

drake7707, 2024-12-26:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Voron 2.4 with toolchanger

### How to reproduce

Add an object, select it to print on the 2nd filament (T1), generate gcode.

There will still be `M104 S280 T0 ; preheat T0 time: 60s` lines even though T0 is not used.

T0 is not even emitted  to PRINT_START: `PRINT_START TOOL_TEMP=280  T1_TEMP=280 T1_IDLE_TEMP=150     BED_TEMP=105 TOOL=1 CHAMBER_TEMP=0 FILAMENT=ABS`

### Actual results

T0 gets heated to print temp even though it isn't used

### Expected results

T0 should not be preheated when it's never used

### Project file & Debug log uploads

[Stealthburner_face_T1.zip](https://github.com/user-attachments/files/18253374/Stealthburner_face_T1.zip)
[debug_Thu_Dec_26_11_36_45_13700.log.zip](https://github.com/user-attachments/files/18253377/debug_Thu_Dec_26_11_36_45_13700.log.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

drake7707, 2024-12-28:
Seems to be because of TIMELAPSE_TAKE_FRAME in the before layer change gcode. The postprocessor sees a line starting with a T and then parses the rest of the string with a cast to 0, so it thinks T0 is called every layer change.

closed 2025-01-22 by Noisyfox

darkzeroat, 2026-02-15:
im having the same issue but not using timelapse so something else is causing it
also happens with 2.3.2-dev

PR #7926 (2025-01-22): Fix issue that `TIMELAPSE_TAKE_FRAME` is recognized as tool change command
… and triggers preheat 

Fix #7842