OrcaSlicer/OrcaSlicer #7848: New Setting: Automatically switch to orthographic mode when using sideview

Minecreator200, 2024-12-27:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I wouldn't say it's a problem, but think that it would be a **great enhancement in user friendliness**.
It would make placing objects way easier as opposed to perspective view.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Whenever the setting is active and all camera angles are a multiple of 90deg switch to orthographic mode.
As already stated above, it would make placing objects way easier.

Pseudo-Code:
```
if (automaticChangeEnabled)
    Renderer.setCameraMode(
        (camera.rotX % 90 == 0 && camera.rotY % 90 == 0 && camera.rotZ % 90 == 0) ?
        CameraMode.ORTHOGRAPHIC :
        CameraMode.PERSPECTIVE
    );
```

buzzhuzz, 2024-12-28:
Having this feature implemented as proposed may result in jerky view since projection might be changed while orbiting build plate.

Better implementation might be "automatically switch to orthographic projection on setting camera view using Ctrl+<1..6>. Switch back to perspective on camera rotation"

Minecreator200, 2024-12-28:
Couldn't we just make it so it only does it if the small cube in the bottom left corner of the view is clicked? That's probably a good compromise.

Noisyfox, 2025-01-05:
> Couldn't we just make it so it only does it if the small cube in the bottom left corner of the view is clicked? That's probably a good compromise.

I remember saw similar behavior in some 3D modeling softwares.

Minecreator200, 2025-01-05:
> > Couldn't we just make it so it only does it if the small cube in the bottom left corner of the view is clicked? That's probably a good compromise.
> 
> I remember saw similar behavior in some 3D modeling softwares, though I can't tell exactly which one.

Yes, Blender does have this feature for example. Some like it, some don't. That's why I had the idea of the toggle switch in the settings.

Noisyfox, 2025-02-04:
Please try https://github.com/SoftFever/OrcaSlicer/pull/8312

Minecreator200, 2025-02-09:
See Noisyfoxes pull request #8312

PR #8312 (2025-03-09): QoL: Add auto perspective
This PR introduces auto perspective that works similar to Blender:

- When click the top/bottom/left/right/front/back on the 3D navigator, or switch to those view from the menu, or by shortcuts (Ctrl+1-6), the camera automatically switch to orthogonal view
- When switch to any other angles, either by dragging with mouse/track pad, or by clicking other non-perpendicular faces on 3D navigator, or select default view from menu/short cut (Ctrl+0), the camera automatically switch to perspective view
- Panning and scalling won't trigger the view change
- This is optional an [... cut]
closed 2025-03-09 by SoftFever