OrcaSlicer/OrcaSlicer #7852: Prevent Color Contamination from Nozzle Blobs in Multi-Color Printing - Add ability to wipe on layer(s) change or at certain points

parrothd, 2024-12-28:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

When performing multi-color printing, blobs of filament can accumulate on the nozzle tip. These blobs may inadvertently get deposited onto the print surface, often carrying remnants of a previous color. This issue can result in color contamination, such as bleed-through or dragging of unintended colors, particularly on features like colored signs or top/bottom layers of multi-color prints. This negatively impacts the overall print quality and appearance.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Provide an option to enable additional nozzle wiping, either automatically on layer changes or manually at user-defined spots. This feature would allow users to select specific locations for extra wiping to prevent color contamination and ensure cleaner multi-color prints.

### Describe alternatives you've considered

I have experimented with various wipers, applying oils, adjusting print speeds, and fine-tuning retraction settings. Unfortunately, none of these approaches effectively remove blobs during the printing process, leaving the issue unresolved.

tlhintoq, 2024-12-29:
Pretty sure you can already do that.
You tick `Retract on layer change` and set `wipe on retraction` in combination with options like `wipe to infil`

If you're getting blobs, that sound like over extrusion or just not retracting.

parrothd, 2024-12-29:
> Pretty sure you can already do that. You tick `Retract on layer change` and set `wipe on retraction` in combination with options like `wipe to infil`
> 
> If you're getting blobs, that sound like over extrusion or just not retracting.

"Wipe on retraction" moves the nozzle along the last extrusion path, it doesn't wipe the nozzle on the wiper to clean it.

"If you're getting blobs, that sound like over extrusion or just not retracting"
yea, unfortunately this really doesn't help with PETG/Abs coloring printing.

parrothd, 2024-12-29:
I updated the title so its not as confusing. :)

tlhintoq, 2024-12-29:
> with PETG/Abs coloring printing, the nozzle actually gets pushed thru the PETG and builds up on the nozzle.

I do 99% PETG printing.  But you lost me there.  "WITH PETG the nozzle gets pushed through the plastic" ?  Are you saying you feel the travels are different for PLA and PETG?  The nozzle gets pushed through the already extruded PETG differently than if you were using PLA?

I mean... I know  guy doing these multi-color prints of Martian space ships.  He's not getting blobs and color bleeding.  Though he did tell me he tuned everything from flow, and material density to the color purging to within an inch of its life to get these results.

[image]

parrothd, 2024-12-29:
That's actually a pretty simple color print, very few color changes and if there's a blob or color defect your probably not going to notice it. I'm doing signs, with white and black lettering and logos where a blob or a color drag across another color wastes the whole print. You should try multiple color prints, especially with petg. 😀

tlhintoq, 2024-12-29:
I appreciate the thought about me looking in to it.  I've been 99% PETG shop since about 2009, so pretty familiar.  I totally get where you're coming from with color bleeds.  I'm from an era when all color changes were manual, pre-dating automatic color changing systems.  So I get it.  It really stinks to have a big smear of color across a field of white.

kylek29, 2025-01-02:
For this to work, they'd probably need to add a Custom G-Code template block to the Machine settings so you can define the wipe procedure, since the machines normally have those in their startup code and every wipe procedure will be different.