OrcaSlicer/OrcaSlicer #7854: Stringing bug report + solution or workaround

newoldschool, 2024-12-28:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS

### OS Version

MacOS sequoia

### Additional system information

Hello I want to report a bug that I was suffering for severals days or weeks I'm not sure for how long.

I've lost a crazy amount of hours searching for solution on internet, trying several settings at every part of the slicer software, and more thank 2kg of filaments wasted on test prints and lost prints for clients due to this issue.

I was suffering of very bad stringing, I knew my settings were at least okay, everything should we working all right.
I tried everything from temperature to retraction length and speed, z-hop, checking a lot of checkboxes I tried everything no kidding.

Until I accidentally changed ONE little setting because I suddenly put 4 models in my plate to test different this...
I've just changed the option "Print sequence: By layer" to "By object".

Just that. And suddenly all the stringing disappeared.
Using the same settings, I even tried other similar settings in retraction and z-hop etc just to see if was a lucky shot, but changing settings was not an issue, the stringing always appear when I select By laver, and the are always gone when I select By Object.
After that I tried with fresh new files, from zero, and different testing models and always the same workaround.

Just to clarify, maybe this was happening before, sometimes I attributed the problem to wet filament, or difficult geometry, I used workarounds or sometimes I just burn the strings with the heatgun. But right now was being really bad stringing, attached you can find to examples, one is the test model, side by side with both printer sequence settings, and the other is an example of a print for a client.

Maybe this is just the tip of the iceberg, and with this you can start to solve that bug.
I really hope you can find a solution for this very soon and give us a new software update.
I've lost so many hours not working and testing settings, and so many filament, the best news I can receive is that you found the solution and you can give us a software update.

Thanks in advance!

### Printer

Bambu Lab A1

### How to reproduce

1. import a stringing test like the one I've uploaded
2. be sure you have select By layer in the plate settings, and do not touch the Process settings, do not change the Print sequence there, just in the plate settings.
3. print
4. after the print is finished, change the plate settings now choose the option By object.
5 .print

### Actual results

With "by layer" option selected, there is stringing, sometimes is not that much, in a lot of cases (in my case) is really bad.
With "by object" option selected, no stringing at all and great quality printing.

### Expected results

"by layer" and "by object" should give the same exact quality print result.

### Project file & Debug log uploads

[Stringing bug report + solution or workaround.zip](https://github.com/user-attachments/files/18268021/Stringing.bug.report.%2B.solution.or.workaround.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

[image]
[image]

tlhintoq, 2024-12-29:
This isn't a bug.  This is the OP starting to wrap their head around 3d printing as something new to them.

> Expected results
"by layer" and "by object" should give the same exact quality print result.

Never going to happen because the mechanics of how the print is being done is different.  "By object" means the nozzle never leaves the object until the object is done.  Therefore there's no dragging of the obviously damp filament away from the object to produce the string.

This is why so many time, in so many social media channels the advice to print "by object" is given over and over and over and over.

The OP is learning something new and starting to grasp how 3d printing works.  That's awesome.  But that doesn't make it a bug in the slicer or something that the mechanics of manufacturing can be magically coded away.

newoldschool, 2024-12-29:
What do you mean by "...means the nozzle never leaves the object until the object is done..." ?

The nozzle leaves the object every layer going from one cone to the other.
How does "by object or by layer" affect the air travel from one cone to the other and the string that is being transported?

Or why in the case of "by object" there is no stringing in the dozens of traveling from one cone to the other?

newoldschool, 2024-12-29:
[image]
in this image you can see what I'm talking about, there are 29 air travels going from one cone to the other.

why the plate option "by layer / by object" affect how much string is pulled going from one cone to the other?

the amount of air travel is the same

tlhintoq, 2024-12-29:
Ohhhh... That clarifies things a bit.
So you're saying that with only 1 object - the stringing test cones - that the `print by` choice made a difference.
Do I have that right?

newoldschool, 2024-12-30:
hello, yes that's right, sorry I'm doing my best with the English I know.

it's so amazing and strange at the same time to see how setting like this can solve the stringing problem.
or maybe we can think the other way and say the selecting the default "by layer" is causing stringing for some reason, because I don't touch any other settings, is like I can turn "stringing" on and off with the "by object" check box

I can make a video too if it's helpful

gregorschulz, 2024-12-31:
Hi,
I have some serious stringing problem which occurred when printing Notre-Dame in big :( but not in the stringing calibration of the Orca Slicer…

So your issue draw my attention and I was hoping for a solution.
I‘ve printed [this test-model](https://makerworld.com/en/models/862372#profileId-812883) in „by Layer“ and „by Object“, but unfortunately I found no differences:

[image]

staal54a, 2025-02-15:
> <img alt="image" width="631" src="https://private-user-images.githubusercontent.com/5025734/399133303-c579c728-0ced-4a40-8d8c-2102eee0564b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mzk2NjA4MzksIm5iZiI6MTczOTY2MDUzOSwicGF0aCI6Ii81MDI1NzM0LzM5OTEzMzMwMy1jNTc5YzcyOC0wY2VkLTRhNDAtOGQ4Yy0yMTAyZWVlMDU2NGIucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDIxNSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAyMTVUMjMwMjE5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NTEyNmI5NzIyM2RlMjZmMTk2OGFkZTMzNmE2ZTQ0ODRkNzQwNjRhMjBhZDllNGUzYzVlYzcxMjk2ZDQyNmUxZSZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.ExfCHpn95tBYRGU9uKfrGcHZopDJ8WBRdXEVS8R1aqo"> in this image you can see what I'm talking about, there are 29 air travels going from one cone to the other.
> why the plate option "by layer / by object" affect how much string is pulled going from one cone to the other?
> 
> the amount of air travel is the same

Is this the screenshot of the `Print By Layer` or the `Print By Object` preview?

closed 2025-05-25 by github-actions[bot]