OrcaSlicer/OrcaSlicer #7859: Add a {scale} placeholder for file name formatting

Venuman22, 2024-12-29:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Currently, Orca Slicer does not provide a way to automatically include the scale of the model in the exported G-code filename. For users who frequently print models at various scales, it becomes challenging to differentiate between files without manually modifying the filename or remembering the scale used for each print.

This is especially problematic when:
	1.	Multiple iterations of the same file are printed at different scales.
	2.	Users manage large projects where scale is a critical parameter for organization.
	3.	Models are printed with minor scale adjustments, which might not be visually apparent in the file name.

Without an automated solution, users must manually rename files or rely on external documentation to keep track of the scale for each print. This process is both time-consuming and prone to human error.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I would like to request a feature to include the scale of the first object (if multiple objects) or the only object (if one object) in the exported G-code filename.
	This feature would be extremely helpful for tracking prints with scaled models, especially when working with multiple iterations of the same file at different scales.
	Example Filename Format:
{input_filename_base}_{scale}%_{printer_model}_{print_time}.gcode
		•	For a 150% scaled object, the filename would show 150% or 1.5x.
	•	If no scaling is applied, it could default to 100% or 1.0x.
	Thank you for considering this enhancement!

### Describe alternatives you've considered

Manual file renaming, keeping track of scale for different files in a separate notebook

tlhintoq, 2024-12-29:
Well... It wouldn't be a single value like `150%` since you can scale all 3 axis independently.
It would have to be something more like `150x_124y_100z`

> Manual file renaming, keeping track of scale for different files in a separate notebook
Or... Just write it as a note on the plate in the slicer.  Either as the plate name, or as a non-printing text element on the plate.  In a notebook sounds like a mess to keep sync'd

Venuman22, 2024-12-30:
Really good point! I guess I was thinking of it more for printable figures, flexible dragons, or other prints that you would usually just scale up or down uniformly which is most of what I print. I print either 70%, 100%, or 150% uniform scale figures often and tend to scale up or down most of the cool prints I find to test overhangs on a smaller scale or just have a smaller version of a cool figure. So I was thinking this addition to the file name would just need to create a placeholder for the scale % for the x axis only and that would effectively be the solution for my feature request since it would be the same scale % for all three if the file was uniformly scaled. But based on your point if each scale percentage had its own placeholder (e.g. x_scale_percentage, y_scale_percentage, z_scale percentage) then a person who does an individual scale for each axis would also be able to edit the file name to include all of that info very easily and it would probably be extremely useful especially for people who make models of varying specific heights and can easily identify which model is the right height for their purpose and set it to print with the filename providing that information automatically.