OrcaSlicer/OrcaSlicer #7872: Non-Planar Z Contouring Integration

Theaninova, 2024-12-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

First things out of the way **I have a working [post-processing script](https://github.com/Theaninova/GCodeZAA)** that already does this, but is a bit cumbersome to work with.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

## What is non-planar z-contouring

I've had this method as a conceptual idea for a long time, but never got around to implementing it, and was equally surprised that no one else has either.

In essence what it does is take the existing sliced lines, subdivide them (I use a 0.1mm resolution right now), and snap them to the actual surface while adjusting flow to vary the layer hight by +/-50% depending on the new z position.

Benefits:
- Improved surface smoothness at shallow angles
- Precise z height for planar surfaces
- Sub-layer detail/surface textures
- Sanding z contoured surfaces seems to yield better results quicker compared to sanding normally printed surfaces

All of this at nearly zero extra print time and in a pretty solid already working implementation.

I suspect there there's a good amount of additional improvements that could be done inside the slicer to further enhance quality.

What I can think of what should probably be tried
- More aggressive ironing that bleeds into the layer below
- Flow rate tuning (right now I just use proportional values)
- Overhang contouring
- Bridge contouring? (this one is probably a bad idea)

### Describe alternatives you've considered

This is not true non-planar slicing, but a big step towards a future where we can have that.

### Additional context

## Results

All of these are sliced with a pretty standard profile, at **0.2mm** layer height and a **0.4mm** nozzle, as well as **ironing** for top surfaces.

All of them are printed on a Voron 2.4 Stealthchanger. I'm hunting down a few print quality issues, but don't let that distract you from the surface finish.

### Europe Map

This one I selected as a torture test, and printed once with z contouring and once plain.
The model is from [Printables](https://www.printables.com/model/401964-3d-relief-europe-topology-map/), with way less polygons and scaled to 75x50x2mm.

During the geometry decimation I messed up some of the flat surfaces, which is actually really interesting because some of these miniscule issues show up in the z contoured print.

<div style="display: flex">
[image]
[image]
</div>

I was a bit surprised how well the plain print turned out, I think it might arguably look better but since I messed up the geometry I think this is mostly down to all those issues being exaggerated.

The z contoured print has a lot more subtle features come through

<div style="display: flex">
[image]
[image]
</div>

### Benchy

This is a split Benchy for demo purposes, **only the left half has z contouring** while the right half is plain.

It also has other print issues that are unrelated to z contouring, _so please ignore them :)_.

#### Railing
Noticeable improvement

<div style="display: flex">
[image]
[image]
</div>

<div style="display: flex">
[image]
[image]
</div>

<div style="display: flex">
[image]
[image]
</div>

#### Roof

Noticeable improvement

<div style="display: flex">
[image]
[image]
[image]
</div>

#### Porthole

Slight reduction in layer lines

<div style="display: flex">
[image]
[image]
[image]
</div>

#### Bow Railing

Negligible difference, as expected

<div style="display: flex">
[image]
[image]
[image]
</div>

#### Floor

I just wanted to show this off because I think it's cool, the floor with z-contouring is 0.1mm lower than the one without.

[image]

xnfra, 2026-02-07:
I really hope this is added to orca as a simple toggle.

skl111, 2026-03-01:
there is a fork of Bambu studio which was featured in CNC Kitchen that does this I think

https://github.com/adob/BambuStudio-ZAA

CNC Kitchen

https://www.youtube.com/watch?v=0Hi9lBjZGIM

This with ironing option could change the surface finish a lot.

Maybe work with that Fork maintainer to implement yours and his ideas if you haven't already started to :)