OrcaSlicer/OrcaSlicer #7873: Odd Prime Tower Issue

giveen, 2024-12-30:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.3.0-dev (and occasccionaly in 2.2.0)

### Operating System (OS)

Windows

### OS Version

10

### Printer

Bambu X1C

### How to reproduce

Occassionally when I add multi color models the prime tower goes off the board and decides it wants to make itself really really big and flat.

### Actual results

[image]

### Expected results

It obviously shouldnt be like that.

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/18275716/log.zip)
[odd_prime_tower.zip](https://github.com/user-attachments/files/18275717/odd_prime_tower.zip)

### Checklist of files to include

- [X] Log file
- [X] Project file

Dropmjr, 2025-01-02:
Same thing has been happening to me, i'm using the BBL A1 combo.  Hope this gets fixed soon

Noisyfox, 2025-01-12:
It's the issue of the "Precise Z height" option.

LehoooJ, 2025-01-23:
Prime tower SIZE doesn't response to settings like Prime Volume no change if 1 or 500mm3

[image]

more over what is my problem that I have double extruder so I need very little purge...but the prime tower minimum size doesnt get smaller. The size of tower is continous to be big as the second extruder minimum size hold about 150mm3 even the 1st can be pusched to about 40mm...... I could need just 10 for clean start of doble extruder.

[image]

LehoooJ, 2025-03-15:
This worked....needed to turn off/ on multi extruder first bevor the other setting enabled to change ....wird but this helped
https://github.com/SoftFever/OrcaSlicer/issues/8120#issuecomment-2609357606

closed 2025-06-21 by github-actions[bot]