OrcaSlicer/OrcaSlicer #7874: Add an z offset calibration test

santiagomoneta, 2024-12-30:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

No

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

I printed a single layer and send a commant to lower the z offset by 0.5... 5 times... at teh end of the print I check and 'rollback' (adding the same value as positivem add an extra 0.05 for aasurance) and SAVE.... then do it again but with 0.025... now I have a great first layer.. would be nice to have a model that prints a rectacle and change the offset mutliple times and then simply pass the value that look best... similar to the YOLO flow calibration... use the CURRENT z offset maybe with .1 extra and go down by X....

### Describe alternatives you've considered

mmm manually sendiong the command in mainsail works... but require me to keep looking and manually sending the command... and I cannot add custom code on the same layer :( ... im not that good editing gcode by hand either...

SoftFever, 2025-03-21:
The issue is that pre-coding the z-offset in G-code can be risky.
It's generally advised to tune this process carefully, one should babysitting the whole process.

I typically slice a cube using 15 loops of brim, then live adjust z offset and observe the brim to determine an optimal z-offset.
While printing, you can manually adjust the z offset on-the-fly, which helps avoid needing multiple print blocks.

JKubovy, 2025-09-07:
I found that the easiest way to calibrate the Z offset is to use this model [printables.com/model/251587-stress-free-first-layer-calibration-in-less-than-5](https://www.printables.com/model/251587-stress-free-first-layer-calibration-in-less-than-5) and slowly adjust the offset after a few segments are printed. You should immediately see the result of the change.