OrcaSlicer/OrcaSlicer #7878: Force first color/filament to print

kzemlin, 2024-12-31:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I am printing some two-color tags.  All tag variants were designed from the same base model.  Only the text was changed.  Orange text on the black tag.  With all 9 tags on the plate it always slices black first.  I can deselect some of the tags, however, and it will then print orange first.  When the tags print orange first, the lettering that is face down on the plate prints much cleaner.  Wide one-layer skirt shows which color is printing first.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

In a multicolor/material print I would like to be able to select which filament will print first.  I have tried all kinds of changes to the models and color painting.  I can not determine what is causing some tags to print black first and some to print orange first.

### Describe alternatives you've considered

I tried changing the base material for the parts vs the painted color - I thought maybe the base color would print first.  That didn't resolve it.
I tried reordering the spools in the AMS.  That did not make a difference.

### Additional context

[image]

[image]

If I print all at once, black prints first.
[image]

superhans65, 2024-12-31:
[image]
Have you tried this?

kzemlin, 2025-01-01:
Bingo.  Thank you.  Had never poked that button.

bistory, 2025-01-04:
> Bingo. Thank you. Had never poked that button.

You should close this issue then ;-)

closed 2025-01-04 by kzemlin