OrcaSlicer/OrcaSlicer #7880: Keep User Presets for process between Printers

soundtechscott, 2025-01-01:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

I shouldnt have to make new profiles for different processes every time I change printers in Orca
we need a way to make the Process Universal and not "Inherited" by the printer chosen

for now my temp workaround is to go to my User Folder
C:\Users\WindowsUserName\AppData\Roaming\OrcaSlicer\user\default\process
and manually edit the json and change the line about Inherits and the profile it chose to 
"inherits": "",

then I just made sure all the settings were adjusted to match what I want it to be and saved it
creating my First preset was tedious
but when I change from .4mm nozzle to a .6mm nozzle or to a .8 nozzle for q quick size test of something, 
the only thing that really has to change most of the time is my Layer height, which can be easily auto calculated by using % instead of absolute MM

when I change from my Neptune 4 Max to my K2 Plus, most of the settings are just fine to keep
so I can use a universal profile instead of one specific to that 1 printer

there are small differences, but I'd rather start with my known good settings for layer heights and infills and only have to adjust the couple of settings that need to be changed and save it as a k2 preset or a N4 preset and be able to start with either if I feel the desire to do so

I already have to change most of them anyways because the system presets suck.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

the solution is to give us a universal preset saving feature, just add an extra check box to the save dialog, and then in the json, set all values instead of just the changed values from the inherited system preset

I already have to change most of them anyways because the system presets suck.
so why make me have to keep a system preset around to create new ones from

Brjodaro, 2025-04-13:
I've been trying to find information about this issue but this thread is all I see. I can't imagine they would actually not let you share process settings across printers, but I see no way to do it. Are they aware that some people own more than one printer??

QuackenAudio, 2025-04-14:
yeah... I just want a checkbox for filaments and printer settings that I can use between multiple printers... global, not locked to the printer

my temporary fix is that I have to manually copy and paste everything to the folder for the other printer...
its dumb...
let me make it global and if I want to save a preset specific to a printer, I'll put it in the name

for example, my Sunlu PLA works the same on all my printers and I literally use the exact same filamet profile and slicer settings on all my other printers with that filament.
same speeds, same flow rates, same temperature, same everything.

no reason to make me have to make a new profile for it with 2 different printers...
Just give me a checkbox,... global, or local to that printer only

also, stop referencing back to other presets in the JSON, make all the settings there in every file.

FigureOfStickman, 2025-05-30:
> my temporary fix is that I have to manually copy and paste everything to the folder for the other printer... its dumb... let me make it global and if I want to save a preset specific to a printer, I'll put it in the name

I guess I'm gonna do that. Do you go line-by-line, copying and pasting the fields? Or is there another way?