OrcaSlicer/OrcaSlicer #7886: Adaptive layer height

zelk12, 2025-01-01:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

The current system does not allow for fast and accurate printing of some of the slanted surfaces.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Change the height of the line layer within a specified range to improve the print quality of slanted surfaces. (I don't know if the following function is available or not at the moment) Make it possible to print perimeters and fills with different layer heights.

### Additional context

This is what the top element of the circle looks like with a layer height of 0,2 mm
[image]

0,1 mm
[image]

0,05 mm
[image]

0,01 mm
[image]

zelk12, 2025-01-01:
Somewhat misspelled, the task is to change the height of the line, not the layer.

Example:
[image]
[image]

VigilantGardener, 2025-01-04:
In the process profile, under strength there is an option called "Infill Combination".  It's near the bottom of the page.

Does that do what you're looking for?

It should let you set layer height to whatever, but the infill will print with taller lines.  Giving, and this is how I describe it, high detail perimeters with a faster infill.

zelk12, 2025-01-04:
Hello, thanks for the reply, yes I did after seeing this, and it fulfils the second part of what you described, but unfortunately not the first.

Noisyfox, 2025-01-12:
#7872 seems related?

zelk12, 2025-01-12:
It's possible.