OrcaSlicer/OrcaSlicer #7893: Zhop on travel

r00tdownx, 2025-01-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Can't Zhop on travel without retraction.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Hi it would be great to have Zhop on travel because a lot of the time I don't want to retract and it would take stress of the extruder cause right now have to force a retract to get a zhop on travel.

igiannakas, 2025-01-03:
Z hop without retraction = stringing. I would not recommend this as the nozzle will still be pressurised and lifting it off the part would create a slight ooze which results in stringing.

ekare, 2025-01-24:
I was looking for the same request. Normal retract can be done per object but travelling between object and supports (and other objects) requires a different setting. Especially travelling between organic supports and the objects's itself results destructive results. 

Normal retraction with z-hop can be enabled/disabled
z-hop (with or without retract) while travelling between supports and other objects may be forced by a distance. 

Even sometimes oozing can be a feature (e.g. while making brush, fur etc.)

IguanaBob, 2025-02-04:
I would find this useful. On my Kobra 3 when printing with 98A TPU (ACE Pro not in use) if I retract at all I find it jams the extruder due to bad design, but I want to use Z-hop to prevent the nozzle scratching a top surface that has to be traveled over as the next layer is completely surrounded by the top surface of the previous layer. Only way I could make this work was have retraction+zhop enabled then edit the retraction lines out of the .gcode file. If I set retraction to 0 it also disables zhop. I would rather deal with a string on this layer change instead of a scratch.

CSV-12, 2025-10-22:
I will try that, I'm in the middle of a big 24 hour 1.7kg print right now, so I cannot change it on the fly.
 @MxBrnr on Discord suggested the same.  The _reduce infill retraction_ was checked, by default (??) and the setting for Z-hop in the filament specific section did not include this (?) or any link to it (?).
This default, and the location of the setting should be grouped where it is most logical for most users, or at least duplicated, if you think the present location is a good idea. I did not even know about this setting, as do many others.
A google search gives many hits on this issue over many years.

The effect of crashing the nozzle is serious.   Googles AI response does not include  _reduce infill retraction_
Also Google _3d print nozzle crashing on infill_

igiannakas, 2025-10-22:
I’ve had zero nozzle crashing on my printer and have more than 10k hours on it. Disable that option.

If retraction over infill is disabled then no z hop will happen (as z hop needs retraction to have happened before.).

If your nozzle is crashing on your infill you need to also check part warpage, and ensure you’re not using overlapping infills like grid…

CSV-12, 2025-10-22:
I will of course uncheck the _reduce infill retraction_.
My question is why is this even not a default state, and why is it not put in a section where it is logical to find it, with the rest of the Z-hop settings?
If you have any connections with the dev's maybe you can inform them of this MAJOR issue.

igiannakas, 2025-10-22:
Because the effect of z hop being ignored is a side effect t / consequence. This setting forces retraction on infill. That is a print profile setting. If you have z hop off, that setting is still valid for example.
Also you can choose to enable it or disable it depending on the LH of your print profile etc vs if it was an extruder setting you wouldn’t be able to as z hop is a filament and extruder setting.

rabbit3dcustom, 2026-09-03:
> Disable reduce infill retraction and report back whether this resolves it. https://github.com/SoftFever/OrcaSlicer/wiki/others_settings_g_code_output#reduce-infill-retraction

Thank you so much!!! 

i was struggling with a defect on the surface of the printed parts in TPU. 
[image]

The problem was on the the "Internal solid infill" (layer), the nozzle keep scratching the surface of the printed parts and that scratch was surfaced on every next layer...

Despite the name of the parameter and the meaning "skip retractions for travel moves that occur entirely inside infill regions", seems that "internal solid infill" is part of the infill and the parameter applies to it. i would like to recommend to update the wiki to specify that "internal solid infill" is part of infill itself.

[image]

Again, Thank you so much for the insight and your time, so appreciate.
Rabbit.