OrcaSlicer/OrcaSlicer #7895: más ajustes de flujo

javier23329, 2025-01-02:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Estimado equipo de OrcaSlicer,

me gustaría compartir con ustedes un problema que he estado experimentando con los laminadores actuales y las limitaciones que estoy enfrentando, con la esperanza de que puedan considerar implementarlo en una futura actualización.

Soy un usuario avanzado de impresión 3D y trabajo con materiales que requieren ajustes muy finos para garantizar la calidad óptima de mis impresiones. Sin embargo, encontró que los laminadores más populares, incluido OrcaSlicer, no ofrecen opciones suficientes para ajustar el flujo de manera precisa y específica para diferentes partes de la impresión, como capas sólidas, rellenos, capas superiores, etc.

Actualmente, la única opción disponible en estos laminadores es el ajuste global de flujo, pero esto no permite optimizar la calidad de la impresión para diferentes tipos de capas. Por ejemplo, las capas superiores suelen mostrar problemas de sobreextrusión debido a la necesidad de un flujo distinto al de las capas de relleno o las capas sólidas inferiores . Esta falta de control más granular limita enormemente la calidad final de las impresiones, ya que no se pueden ajustar los parámetros de flujo según las diferentes necesidades de cada sección de la pieza impresa.

En Creality Slicer , hace algún tiempo, existía una opción para ajustar estos parámetros de manera independiente por tipo de capa, y encontré este nivel de control muy útil. Sin embargo, con la migración a laminadores más modernos como OrcaSlicer, ha notado que estas opciones avanzadas han desaparecido, y esto se ha convertido en una limitación significativa para lograr impresiones de la más alta calidad.

Dado que OrcaSlicer es un cortador de código abierto, me gustaría preguntar si sería posible considerar la inclusión de un sistema de ajuste de flujo independiente por tipo de capa. Esto permitiría a los usuarios configurar el flujo específicamente para capas superiores , relleno , capas sólidas y otros aspectos, mejorando así la calidad general de las impresiones y permitiendo una personalización mucho mayor.

Además, creo que la inclusión de estas opciones avanzadas podría ofrecer una ventaja competitiva significativa para OrcaSlicer en el mercado de laminadores. Mientras que otras cortadoras populares, como PrusaSlicer y Bambu Studio, siguen ofreciendo solo ajustes limitados en términos de flujo, un control detallado sobre cada capa podría atraer a usuarios más avanzados y profesionales que buscan una personalización precisa y una calidad de impresión excepcional. Este tipo de funcionalidad sería un gran diferenciador y podría posicionar a OrcaSlicer como una herramienta más poderosa y flexible para usuarios exigentes.

Gracias por su tiempo y por el excelente trabajo que están haciendo con OrcaSlicer. Espero con interés cualquier respuesta y actualización sobre este tema.

Atentamente,
Javier Gimeno ( Universo XYZ)

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Añadir los ajustes antes mencionados.

### Additional context

[image]

pedroloco2, 2025-01-02:
Podes hacerlo utilizando modificadores. Saludos.

javier23329, 2025-01-03:
no se puede establecer el flujo de las capas solidas con modificadores, le agradecería que no proporcione información errónea, y en el caso de que sea cierto proporcione más información. Gracias.

DVSVIDEO, 2025-01-03:
> Gracias por la respuesta. He revisado detenidamente las opciones disponibles en los modificadores, pero no he encontrado una forma de ajustar el flujo específicamente para las capas sólidas u otras partes de la impresión. Si existe esta posibilidad, le agradecería mucho que pudiera detallar los pasos o proporcionar más información al respecto. En caso contrario, sería ideal aclarar esta limitación para evitar confusiones futuras.

Usted menciona en la hipótesis que no se puede ajustar el flujo de la capa superior, lo que conduce a la sobreextrusión.
Pues bien, ajustar el flujo de las capas inferior y superior forma parte de Orcaslicer desde hace mucho tiempo.

[image]

DVSVIDEO, 2025-01-03:
There is a setting in the "Quality" menu for "Print line width"
This is just to change the amount of flow.
If you think about it...when you increase the flow for a given layer height, the excess material can only spread laterally (previous layer at the bottom, nozzle tip at the top). This is exactly what increases the line width.
Well you can tune this in the setting mentioned earlier, element by element...
Before suggesting features, reporting errors, do your research carefully.

pedroloco2, 2025-02-02:
[image]