OrcaSlicer/OrcaSlicer #7902: Update wxWidgets to newest version

manio, 2025-01-03:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

ArchLinux

### Printer

Not important

### How to reproduce

Start a program

### Actual results

When using on HiDPI display with scaling enabled (200% in my case) it looks small and upscaled with ugly fonts and small workspace.
Here is a comparition with native wayland app:
[image]

I reported the problem to wxWidgets (https://github.com/wxWidgets/wxWidgets/issues/25038) but the conclusion is that this has been fixed upstream but the OrcaSlicer is still using the old forked version (3.1.5).

This is how is looking OrcaSlider GUI versus new (3.3.0) wxWidgets `minimal` test application:
[image]

### Expected results

Update wxWidgets fork to recent version

### Project file & Debug log uploads

Not applicable

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

This may also help @phrxmd reported issue in #7050. I am not sure but it's worth to try...

buzzhuzz, 2025-01-12:
Hi, could you please test if debug build from https://github.com/SoftFever/OrcaSlicer/pull/8016 will fix your issue?

manio, 2025-01-12:
@buzzhuzz 
I am building it on Arch with [modified PKGBUILD](https://pastebin.com/nCSCjUZK).
Finally it is unfortunately throwing linking error:
```
/usr/bin/ld: /tmp/ccESS9ZG.ltrans109.ltrans.o: undefined reference to symbol 'png_set_read_fn@@PNG16_0'
/usr/bin/ld: /usr/lib/libpng16.so.16: error adding symbols: DSO missing from command line
collect2: error: ld returned 1 exit status
make[2]: *** [src/CMakeFiles/OrcaSlicer.dir/build.make:254: src/orca-slicer] Error 1
make[1]: *** [CMakeFiles/Makefile2:711: src/CMakeFiles/OrcaSlicer.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
==> ERROR: A failure occurred in build().
    Aborting...
```
Maybe I am doing something wrong... but this is rather Arch specific thing. Maybe Maciek (@c2h5oh) can take a look what I am doing wrong. Here's a diff against AUR's PKGBUILD:
```diff
diff --git a/PKGBUILD b/PKGBUILD
index c28436a..d89bd0b 100644
--- a/PKGBUILD
+++ b/PKGBUILD
@@ -13,13 +13,13 @@ makedepends=('cmake' 'extra-cmake-modules' 'git' 'm4' 'pkgconf')
 provides=("orca-slicer")
 conflicts=("orca-slicer")
 source=(
-  "https://github.com/SoftFever/OrcaSlicer/archive/refs/tags/v${_tag}.tar.gz"
+  "https://github.com/buzzhuzz/OrcaSlicer/archive/refs/heads/test/wxwidgets-update.zip"
   "https://raw.githubusercontent.com/SoftFever/OrcaSlicer/v${_tag}/deps/Boost/0001-Boost-fix.patch"
   'CMakeLists.txt.patch'
   'orca-slicer.sh'
   '0002-clipper.patch'
   )
-sha256sums=('90d78d8bd60fb98a4e63708b70e53c3095fb61c16381d36d0063a6d7943a4ec7'
+sha256sums=('4d1198460cee72b742148052a3b7fe578676f3d309d9dcdcf712b0fc765f0a6f'
             'bb2662d0a4c58c43726ec98ef4acf201fcf98719c9bbfd207e2d6cdf695a2093'
             '01171a77d533584026f113092a6586e28c9d87e10117c0f81cb4357d11a29fb1'
             '8c958ecc5522fd00b9e4161e20de438265fd7f31abdbcc63d7e5d90596ff7a19'
@@ -27,7 +27,7 @@ sha256sums=('90d78d8bd60fb98a4e63708b70e53c3095fb61c16381d36d0063a6d7943a4ec7'
 
 prepare() {
   # link up directory
-  ln -sf OrcaSlicer-${_tag} OrcaSlicer
+  ln -sf OrcaSlicer-test-wxwidgets-update OrcaSlicer
   # add missing 0001-Boost-fix.patch
   cp 0001-Boost-fix.patch OrcaSlicer/deps/Boost
   patch -p0 < "$srcdir/CMakeLists.txt.patch"
```
Anyway - I realized that you have a CI attached and found ubuntu artifact: `OrcaSlicer_Linux_ubuntu_2404_PR-8016.zip`, maybe I'll do some use of this...

manio, 2025-01-12:
@buzzhuzz 
Update: OK, I tested the AppImage version. Unfortunately it still doesn't look right comparing to native wxWidgets 3.3 applications:
[image]
The only thing that comes to my mind is wxWidgets 3.3 (tested by me and looking OK) instead of 3.2.6

buzzhuzz, 2025-01-13:
@manio , would you mind testing appimage here https://github.com/buzzhuzz/OrcaSlicer/actions/runs/12751039210 ?

It is a draft of wxWidgets 3.3 support in Orca. It is not fully working (for example, switching to by-object printing causes crash, checkboxes are crooked), but basic functions are working.

However, I don't think this would help with your problem since OrcaSlicer are messing with widgets themes and fonts.  For the fonts rendering issue I'd recommend you test my PRs: https://github.com/SoftFever/OrcaSlicer/pull/7412 and https://github.com/SoftFever/OrcaSlicer/pull/7409

As for the widgets styling and themes: I personally would prefer default system-wide theme like in Prusa Slicer, but I can't justify effort needed for this.

manio, 2025-01-14:
@buzzhuzz 
you was right, version compiled against wxWidgets 3.3 also doesn't look right (compared to native `minimal` app from wxWidgets itself):
[image]

> However, I don't think this would help with your problem since OrcaSlicer are messing with widgets themes and fonts. For the fonts rendering issue I'd recommend you test my PRs: https://github.com/SoftFever/OrcaSlicer/pull/7412 and https://github.com/SoftFever/OrcaSlicer/pull/7409

I can understand that, but I am calling the default "status window" which I believe is totally generated on wxWidgets side.
I am pressing Ctrl+Alt and middle clicking on any place where handler is not registered (https://github.com/wxWidgets/wxWidgets/issues/25038#issuecomment-2567504735):
[image]

If we have the version in pair, maybe this is something OrcaSlicer specific, like [creating window](https://docs.wxwidgets.org/3.2/overview_high_dpi.html)...

Anyway, I'll test the two PRs you mentioned...

buzzhuzz, 2025-02-01:
I just triggered build actions for the changes in mentioned PRs on my fork. Please check https://github.com/buzzhuzz/OrcaSlicer/actions/runs/13084497574 and https://github.com/buzzhuzz/OrcaSlicer/actions/runs/13084499001

As for the current issue report, it may be related to the fact that Orca forces `GDK_BACKEND=x11` which is known for poor UI scaling. I just pushed test change to my repo which allows use of Wayland, you can find test build at k https://github.com/buzzhuzz/OrcaSlicer/actions/runs/13084550066 NOTE that 3d view in this change is broken: some work need to be done to correctly render 3D scenes (prepare, preview) under wayland.

manio, 2025-02-01:
> it may be related to the fact that Orca forces GDK_BACKEND=x11

Oh my ... I was not aware of this! Definitely it was the problems as it was not using wayland at all!
I started testing the assets from the last one and ... boom, here is the comparison side-by-side:

[image]

[image]

And compare the model resolution and font quality for the same tile:

[image]

It would be great to be able to freely scale or disable scaling completely of the whole UI because it is still far too big for me - I'd like to use the whole 4K display workspace, but at least the font and model quality is really great as it should be with proper (not blurry and upscaled) resolution!

Can you fix this upstream?

I'll check the other first two assets later today...

manio, 2025-02-01:
Ok... now testing the first asset:

[image]

Doesn't differ too much to my base ("stock") version, but I can see a small difference eg in this warning message font:

[image]

---
And the second asset comparison:

[image]

Cannot see any difference at the first sight...

nagisa, 2025-05-11:
Could we start by landing a diff along the lines of

```diff
-    ::setenv("GDK_BACKEND", "x11", /* replace */ true);
+    ::setenv("GDK_BACKEND", "x11", /* replace */ false);
```

so that users who have no issues with the wayland backend could opt-in at least?

manio, 2025-08-10:
Yeah, any news on this?

viesturz, 2026-01-31:
Bump as well, this causes Orca to ignore Desktop environment file dialogs, #12125

DeepReef11, 2026-02-04:
Bump, I need this.

In the mean time I will use lf and yank path then in gtk picker, use ctrl+l, paste path. Should work decently