OrcaSlicer/OrcaSlicer #7905: Adaptive speed

zelk12, 2025-01-03:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

For elements with different radii, the optimum print speed may be different, but right now there is only perimeter and small perimeter to choose from.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add a function similar to adaptive pressure advance only by ratio, item size, speed.

DVSVIDEO, 2025-01-03:
There are more and more requests and error reports that are generated out of ignorance of Orcaslicer, or lack of printing knowledge, or laziness....

igiannakas, 2025-01-03:
To be fair, we could do more work on the wiki (which has started but needs a pretty significant time investment to get it done).

zelk12, 2025-01-03:
Just the idea of making it similar to an adaptive pressure advance.

[image]

Only for element size - speed, as an example:
2 mm - 10 mm/s
5 mm - 20 mm/s
8 mm - 30 mm/s

zelk12, 2025-01-03:
I don't know if it's implemented or not, but ideally, an element stability detection system would be nice.  And speed changes depending on stability.

igiannakas, 2025-01-03:
> I don't know if it's implemented or not, but ideally, an element stability detection system would be nice. And speed changes depending on stability.

Ok that’s an interesting idea - the stability element mentioned can be approximated by the feature circumference which is what already exists above. However the current feature only sets a static speed. What you’re proposing I think is a dynamic speed going faster the larger the feature is.

DVSVIDEO, 2025-01-03:
> I don't know if it's implemented or not, but ideally, an element stability detection system would be nice. And speed changes depending on stability.

If you set the acceleration to low 150-300 mm/s2, it will be slow in small areas, even if the speed is set to 300mm/s. It only reaches maximum speed for long lines. This is a dynamic speed control
If stability is the key and not high speed printing, why not use this?

zelk12, 2025-01-05:
I mainly use tree organic supports which I print very thin, now for some models I have to manually set the overall print speed of the supports to 25mm/s and sometimes 10mm/s so the printer doesn't break them. And an adaptive speed system might allow the printer speed to be reduced depending on the "stability" of the element, meaning that stable elements will print fast, and only non-stable elements will print slowly.

DVSVIDEO, 2025-01-15:
> I mainly use tree organic supports which I print very thin, now for some models I have to manually set the overall print speed of the supports to 25mm/s and sometimes 10mm/s so the printer doesn't [...]

So there is a solution! I think the developers are either working on adding new options (non planar upper layer) or fixing existing bugs. What can be solved with a little extra work from the user....is not exactly a priority. Maybe one day... if there is a mass demand for it....

thisnyjared, 2025-01-26:
I think adaptive speed like adaptive layer height would be nice.