OrcaSlicer/OrcaSlicer #7915: Infill speed influences wall speed

dzid26, 2025-01-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Win11

### Printer

SV07+

### How to reproduce

1. open XYZ.3mf project
2. Set Sparse Infill Speed to 20 and slice
3. Set Sparse infill speed to 30  and slice

Observe dramatic difference in sliced _wall_ speed on the color map 

### Actual results

Sparse infill is set below ~20mm/s, the outer wall speed is 200mm/s:
[image]
When sparse infill is above ~30 mm/s, the wall speed is about 70mm/s, except a layer that has a bridge:
[image]

### Expected results

Wall speed should not be influenced by infill speed settings. 

### Project file & Debug log uploads

[XYZ.zip](https://github.com/user-attachments/files/18304949/XYZ.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

buzzhuzz, 2025-01-04:
This should be related to the min layer time.

There are issue closed by timeout https://github.com/SoftFever/OrcaSlicer/issues/4191 which points to the non-obviousness of that setting.

There another PR https://github.com/SoftFever/OrcaSlicer/pull/5148 which addresses the same issue.

dzid26, 2025-01-04:
Correct answer. 
I tentatively suspected something like this, but it didn't occur to me (never printed that fast before).

Layer time can be seen here:
[image]

The setting is in filament options:
[image]

The settings is perfectly justified and below there are options to control its behavior.

closed 2025-01-04 by dzid26