OrcaSlicer/OrcaSlicer #7917: Objects on slicer plate move after save and reopen (Re-Issue of #4016)

yomarcyo, 2025-01-04:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

macOS, Windows

### OS Version

macOS Sequoia, Windows 10 + 11

### Additional system information

CPU: 
Apple Silicon M3
Intel i7 14700k

### Printer

Bambu A1, X1C 

### How to reproduce

1. Use more than one Plate
2. Position multiple objects on each plate
3. Save Project
4. Re-open project

### Actual results

All objects have moved slightly in the Y-axis and left the plate.
[image]

### Expected results

The objects should stay in place between openings.

### Project file & Debug log uploads

Please let me know if this in neccesary. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Issue has been submitted as "[Objects on slicer plate move after save and reopen #4016](https://github.com/SoftFever/OrcaSlicer/issues/4016)" but was never fixed. Other users experience the same problem.

tlhintoq, 2025-01-11:
It isn't just from switching beds/machines in a job.
This job exists only for the Elegoo Giga.  Has only ever been worked with the same Giga as the selected printer.
Open the job... work it... make a change whatever... export... save... new...
And I only change printers after I `NEW` on the current printer.  That way there's never a job changing machines.
And yet... I open the job this morning and a part that belongs on plate 2... was on plate 2 when saved... now is off in mid space between plates 1 and 2.

[image]

closed 2025-01-25 by SoftFever

FelixBuehler, 2025-02-26:
Still not fixed to this day.

darunkle, 2025-02-28:
Yea, still not fixed and does not need to have profiles or bed sizes changed.

tlhintoq, 2025-03-06:
As in previous comment - no change in machine type.  
Re-opened this job simply to print another batch and you can see how far down all the parts are moved.
The batch was centered on bed previous.
All the plates in this job have moved the same amount, whether there was a non-printing text on the bed or not (before someone brings that up)

[image]

FelixBuehler, 2025-03-06:
I just realized, it's fixed in the v2.3.0-beta Release:

https://github.com/SoftFever/OrcaSlicer/pull/7923

darunkle, 2025-03-07:
I just opened a project file in 2.3.0 beta and all but 1 plate was deleted (parts are still there) and they're all shifted - so even when I add new plates, they'll all off.

[image]