OrcaSlicer/OrcaSlicer #7925: Orcaslicer crashing slicing file

mywereld, 2025-01-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0 and 2.3.0-dev

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

CPU: AMD Ryzen 7600X
GPU: Radeon RX 5700 XT
Memory: 32GB

### Printer

Creality Ender 3v3 KE

### How to reproduce

1. Open any STL, 3mf file
2. Choose slice
3. Orca crashes

I have installed 2.2.0 and 2.3.0-dev.
Uninstalled all AMD chipset drivers and reinstalled drivers, no change.

### Actual results

[https://youtu.be/PtWKLkbIVm4](url)

### Expected results

Slicer the project without crashing.

### Project file & Debug log uploads

[crash_Sun_Jan_05_13_34_29_0.log](https://github.com/user-attachments/files/18310047/crash_Sun_Jan_05_13_34_29_0.log)

### Checklist of files to include

- [X] Log file
- [ ] Project file

Noisyfox, 2025-01-05:
Could you provide a 3mf file that could reproduce this issue?

mywereld, 2025-01-05:
Any file, I tried multiple files:
[xyzCalibration_cube.zip](https://github.com/user-attachments/files/18310099/xyzCalibration_cube.zip)

Noisyfox, 2025-01-05:
I need the 3mf file which contains the slicing configs.

mywereld, 2025-01-05:
Sorry, thank you trying to help.
I just loaded the cube and saved the project, cannot even attempt to slice since Orca crashes immediately:
[xyzCalibration_cube (2).zip](https://github.com/user-attachments/files/18310195/xyzCalibration_cube.2.zip)

Noisyfox, 2025-01-05:
What graphic card are you using? Maybe try upgrading the GPU driver?

mywereld, 2025-01-05:
GPU: Radeon RX 5700 XT

I uninstalled all AMD chipset drivers and reinstalled drivers, no change. Also did the same with the GPU drivers.
After uninstalling the GPU drivers I tried to slice with Orca again, same result.

mywereld, 2025-01-09:
I will have to reopen this bug report.
Setting Orca to low priority suddenly does not work anymore. I have to revert to other slicers to get work done.
Attached the latest log, but it pretty much stayed the same:
`Exceptions:
Exception Code :c0000005 ACCESS_VIOLATION
Exception Flag :0x0 NumberParameters :2 
Param 0 :0x0 
Param 1 :0x248 
Context :000000C08A4FECC0 
ContextFlag : 0x10005f, EFlags: 0x10206 
Fault address:  0x2CCE9486 0x1:0xD68486 C:\WINDOWS\System32\DriverStore\FileRepository\u0410212.inf_amd64_daae2c8b5eb35aaa\B409877\atio6axx.dll

Show CallStack:
7ffb2cce9486:DrvSetCallbackProcs [Unknown file][-1]
7ffb2cc7c666:DrvSetCallbackProcs [Unknown file][-1]
7ffb2cda9a0f:DrvSetCallbackProcs [Unknown file][-1]
7ffb2ca61b0d:DllMain [Unknown file][-1]
7ffb2ca62f3f:DllMain [Unknown file][-1]
7ffb2cf89767:eglGetProcAddress [Unknown file][-1]
7ffbf408e8d7:BaseThreadInitThunk [Unknown file][-1]
7ffbf48ffbcc:RtlUserThreadStart [Unknown file][-1]

Logical Address:
0x2CCE9486 0x1:0xD68486 C:\WINDOWS\System32\DriverStore\FileRepository\u0410212.inf_amd64_daae2c8b5eb35aaa\B409877\atio6axx.dll
0x2CC7C666 0x1:0xCFB666 C:\WINDOWS\System32\DriverStore\FileRepository\u0410212.inf_amd64_daae2c8b5eb35aaa\B409877\atio6axx.dll
0x2CDA9A0F 0x1:0xE28A0F C:\WINDOWS\System32\DriverStore\FileRepository\u0410212.inf_amd64_daae2c8b5eb35aaa\B409877\atio6axx.dll
0x2CA61B0D 0x1:0xAE0B0D C:\WINDOWS\System32\DriverStore\FileRepository\u0410212.inf_amd64_daae2c8b5eb35aaa\B409877\atio6axx.dll
0x2CA62F3F 0x1:0xAE1F3F C:\WINDOWS\System32\DriverStore\FileRepository\u0410212.inf_amd64_daae2c8b5eb35aaa\B409877\atio6axx.dll
0x2CF89767 0x1:0x1008767 C:\WINDOWS\System32\DriverStore\FileRepository\u0410212.inf_amd64_daae2c8b5eb35aaa\B409877\atio6axx.dll
0xF408E8D7 0x1:0x2D8D7 C:\WINDOWS\System32\KERNEL32.DLL
0xF48FFBCC 0x1:0xDEBCC C:\WINDOWS\SYSTEM32\ntdll.dll
`

closed 2025-04-17 by github-actions[bot]