OrcaSlicer/OrcaSlicer #7928: Overhangs without supports

zelk12, 2025-01-05:
### Is there an existing issue for this feature request?

- [X] I have searched the existing issues

### Is your feature request related to a problem?

Supports are now built even for items that can be printed without using supports.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Note: This refers to tree-based organic supports.

Customisable system for detecting printable overhangs. In case an overhang is considered printable, exclude this element from the construction of supports.

One of the simplest ways is to check the distance to walls and other supporting elements, a parameter that allows you to set a minimum allowable distance.

There is already a system for detecting overhangs based on angles, you could add a similar system, but take the angle change as a step, or something similar.

Also the overall system for placing supports could be optimised, when placing supports, I'm not sure how supports are placed now, but I've noticed that they go around corners, in this case it could be made to place supports in corners to improve the quality of their printing.

*** Translated with www.DeepL.com/Translator (free version) ***

ShaneDelmore, 2025-01-07:
Can you explain how this differs from support "threshold angle"?

zelk12, 2025-01-07:
Orange is the perimeter, black is the corner designation, and green is the supports.
This is how the angle is conventionally checked now.
[image]

The supports will be constructed as follows.
[image]

The idea, check the corners roughly as follows.
[image]

neo-jayfeather, 2025-01-08:
Your third image wouldn't work, overhangs are not based on the angle relative to the wall, but the angle relative to extrusion. This is why 4 or 5 axis printers can print overhangs without supports. The printer would still struggle with the 3rd image you provided, maybe not in the second section, but the third section would be very difficult, and if your printer could print that, then you could increase your threshold angle to that angle (relative to an axis perpendicular to the bed), and it would generate supports that way. However, this would still create immense overhangs and could and would only work if there was a surface to bridge between.

zelk12, 2025-01-09:
Как я понимаю, многое, в том числе зависит от расстояний к примеру, я могу нормально печатать с углом в 45°, при этом если угол 60° я могу напечатать элемент длинно в 1 см и так далее. Но здесь еще вот какой вопрос, печать арок.

То есть я могу этот элемент напечатать без поддержек.
[image]

Но этот элемент не могу.
[image]

А этот элемент могу, но на меньшее расстояние.
[image]

При этом если посмотреть на эти поддержки, можно заметить, что они расставлены не оптимально.
[image]

[image]

[image]