OrcaSlicer/OrcaSlicer #7929: Skirt height is enforced to 1 for By object printing with no option to disable

maxim-kukushkin, 2025-01-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.1.1

### Operating System (OS)

Linux

### OS Version

Arch Linux

### Printer

Voron Trident

### How to reproduce

Select Others -> Special Mode -> Print Sequence -> By object
Try to set Others -> Skirt -> Skirt height to anything above 1

### Actual results

The below warning message appears and the skirt height is forcibly reset to 1
[image]

### Expected results

As the message states, there's a **chance** of collision, not a **guarantee**
The message must be just a warning, maybe an exclamation mark next to the setting. It must not enforce any settings.

The user knows their printer better, the user sees how it prints and the user knows better which settings are acceptable.

In my case it was printing like this before, I didn't have any collisions. I also know the clearance of my nozzle enough to judge. I'm also experienced enough to check the sliced view and evaluate.

The slicer must not be opinionated, especially as there's no way for it to protect the printing from everything. There're many other ways to perform incorrect settings and ruin the printing. This one is not even fatal (what will happen even if the nozzle collides with a 2-3 layer skirt? Most probably nothing).

### Project file & Debug log uploads

[klein-bottle_v2_small.zip](https://github.com/user-attachments/files/18311318/klein-bottle_v2_small.zip)
[debug_Sat_Jan_04_21_55_12_2487.log.0.zip](https://github.com/user-attachments/files/18311319/debug_Sat_Jan_04_21_55_12_2487.log.0.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

The reason this is reported as a bug, is because it's incorrect that the slicer is selectively opinionated about a specific setting and doesn't let the user decide
It also prevents user form performing correct setting which work for them

maxim-kukushkin, 2025-01-05:
Additionally, it also enforces skirt height with print By Object, even if the objects are far away from each other and even if there's only one object on the build plate

Noisyfox, 2025-01-08:
I guess that's something from the early era when we don't have per-object skirt.

Noisyfox, 2025-01-08:
And this should have already been fixed in 2.2

maxim-kukushkin, 2025-01-11:
Yes, it indeed was fixed in 2.2. I've checked all combinations of skirt kind, height and print sequence.

Sorry for the noise!

closed 2025-01-11 by maxim-kukushkin