OrcaSlicer/OrcaSlicer #7931: "Bottom interface spacing" option has no effect/"Top interface spacing" changes bottom spacing

organicunity, 2025-01-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

i7

### Printer

Creality K1C

### How to reproduce

This is a repost of an existing bug that was not fixed.  See here: https://github.com/SoftFever/OrcaSlicer/issues/2745

Load a [support interface test](https://www.printables.com/model/198420-support-test) model.
    Enable supports/support interface
    Set different values for Top interface spacing and Bottom interface spacing
    Slice model

### Actual results

Both top and bottom interface spacing are the same (mirroring the Top interface spacing value). Top interface spacing affecting both top and bottom can be verified by changing only the Top interface spacing and re-slicing.

### Expected results

    Bottom interface spacing should change the bottom interface spacing. It currently has no discernible effect on slicing.
    Top interface spacing shouldn't change the bottom interface spacing.

### Project file & Debug log uploads

no project or log file needed.  Easy to reproduce bug.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

This is a repost of an existing bug that was not fixed.  See here: https://github.com/SoftFever/OrcaSlicer/issues/2745

This is an essential feature to produce useable build support separation from the model.

Noisyfox, 2025-01-09:
The option you are looking at is called "Support/object xy distance"

organicunity, 2025-01-10:
> The option you are looking for is called "Support/object xy distance"

nope.  "Support/object xy distance" only adjusts the support distance after the first layer.  I need to change it for the initial (interface) first layer.

organicunity, 2025-01-12:
> Though due to the first layer expansion the gap will be smaller than the set value. You'll have to reduce that value too if needed:
> 
> ![image](https://private-user-images.githubusercontent.com/1537155/401919520-a250642a-526e-431a-8723-7c5eee49a6bc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3MjA4NzQsIm5iZiI6MTczNjcyMDU3NCwicGF0aCI6Ii8xNTM3MTU1LzQwMTkxOTUyMC1hMjUwNjQyYS01MjZlLTQzMWEtODcyMy03YzVlZWU0OWE2YmMucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI1MDExMiUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNTAxMTJUMjIyMjU0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9NDBmNzBjNTQ4M2UwMzVkYTIzZGI4ZGM3MmZkY2I4OWFkNTAzYTUwODhkZjU0MWFiNGRjNzE5ZWI4N2Y4ZGNjMyZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.xtjxfWhgphA1aqeHxTcI413Zl2SdLF3RZIOvCafyiW4)

Wow, thank you so much!!
The value didn't chance until > 2mm 
and thats because the "first layer expansion" setting.  So that setting and "object x/y distance" are dependant on each other, if changing the first layer!  Makes sense.

I changed "first layer expansion" to 0 and no "object x/y distance" for support material works as expected.  **Thank you!!!**

organicunity, 2025-01-12:
> The option you are looking for is called "Support/object xy distance"

thanks, updated the bug report to reflect this!

closed 2025-01-12 by organicunity