OrcaSlicer/OrcaSlicer #7934: Bambu TPU for AMS not reconized

Scoutcreed, 2025-01-05:
### Is there an existing issue for this problem?

- [X] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11 pro

### Additional system information

Processor	AMD Ryzen 9 9900X 12-Core Processor               4.40 GHz
Installed RAM	64.0 GB (61.6 GB usable)
GPU: Zotac GeForce RTX 4080 Super 
System type	64-bit operating system, x64-based processor

### Printer

Bambu P1S with standard AMS

### How to reproduce

1. Load TPU for AMS
2. Open Orca Slicer and have it resync AMS.

If you try Bambu studio or Orca 2.3.0 Developer version this will be fixed on power restart of printer.
However if Orca 2.2.0 is opened it breaks the view on all other versions.

### Actual results

? shows up in the slot if TPU is loaded

### Expected results

TPU for AMS should Appear in the slot

### Project file & Debug log uploads

[Project I was working on that had said issue.zip](https://github.com/user-attachments/files/18312108/Project.I.was.working.on.that.had.said.issue.zip)

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Anything else?

I was just trying to create a rubber foot on a support brace using a the Bambu TPU for AMS as the rubber foot in AMS slot 1 and Bambu's sparkle PLA in AMS slot 3. Something I tried but was unsuccessful was I took the profiles from the developer 2.3.0 version and uploaded specifically the files for the P1P into the Bambu Filament file in the 2.2.0 version. This failed probably because I didn't know what I was doing more than the files them self. This seems to be an easy fix if a new stable version is released, but I would recommend instead of having all of the additional functionality of 2.3.0, just to create a 2.2.1 version or something with all of these new filaments ported over.

fenhelix, 2025-01-13:
Can also confirm this issue with orca, yet it works in bambu slicer. 
Had to close orca slicer and open bambu slicer to make this error go away. 
[image]

Math3w, 2025-01-20:
I also confirm the same problems both in Orca and on touchscreen. X1C with TPU for AMS

ondrovic, 2025-04-26:
how is is stale when we are waiting for a fix?

ondrovic, 2025-07-27:
bump still not stale

closed 2025-11-04 by github-actions[bot]